jeremiah-c-leary/vhdl-style-guide #375: Unify 'end keyword' case rules approach.

m-kru, 2020-04-06:
There are some rules that check the case of `end keyword` constructs. However sometimes there is single rule for both word, while sometimes there 2 rules per word. Examples: 
[image]
[image]

I propose to either use single rule everywhere or 2 rules.

m-kru, 2020-06-18:
@jeremiah-c-leary Should we have 2 separate rules for `end` and `keyword` or one rule for `end keyword`?

m-kru, 2020-06-19:
Ok, we are going to have 2 separate rules, as this gives more flexibility.
Actions needed:
- [ ] split architecture_009
- [ ] add rules for checking case of `end function`
- [ ] split if_028
- [ ] split package_006
- [ ] add rules for checking case of `end procedure`

jeremiah-c-leary, 2020-06-21:
I agree with splitting them.  There was a time when I was just trying to go as fast a possible to get rules in without taking the time to really think about the overall strategy.

m-kru, 2020-06-23:
PR is waiting https://github.com/jeremiah-c-leary/vhdl-style-guide/pull/430

closed 2020-07-18 by jeremiah-c-leary