jeremiah-c-leary/vhdl-style-guide #380: Syntax Error after running VSG

HalfInner, 2020-04-28:
**Environment**
VHDL Style Guide (VSG) version 1.2.0

**Describe the bug**
Two problems: 
 * Major: When the procedure is the latest element in architecture group initialization - i'm not sure if I shall call it like this - vsg threat 'end procedure' as a name and change it to end architecture procedure, which is **`Syntax Error`**
 * Minor: Wrongly interpreted indents in procudure

[image]

[image]

**To Reproduce**
As on the print screens:
1. Make procedure as last block
2. Run the script with a fix

**Expected behavior**
Properly adjust code.

**Screenshots**

**Additional context**
Problematic code is on my github /SelomoVHDL

jeremiah-c-leary, 2020-04-30:
greetings @HalfInner ,  I will take a look at this

jeremiah-c-leary, 2020-04-30:
Hey @HalfInner , just wanted to let you know I fixed the issue.  VSG was classifying the 'end loop' as an  'end procedure', which then threw all the indenting off and caused the syntax error.

I released version 1.4.0 with the fix.  Please try it and let me know how it goes.

I will close this ticket in a couple of days if I do not hear back from you.

HalfInner, 2020-05-02:
Hello @jeremiah-c-leary , 
I've tested with 1.5.0 and it works well. Good to hear!
Thanks for fix.

closed 2020-05-02 by HalfInner