jeremiah-c-leary/vhdl-style-guide #383: Nested for loop indentation

tlbeng, 2020-05-04:
**Environment**
Windows 10, Python 3.7.7, vsg 1.5.0

**Describe the bug**
VSG incorrectly reports invalid indentation at the end of nested for loops

**To Reproduce**
run `vsg -f nested_loop.vhd` for a file with the following contents:

~~~vhdl
-- Nested Loop example

library ieee;
  use ieee.std_logic_1164.all;

entity NESTED_LOOP is
  port (
    CLK             : in    std_logic;
    DATA            : out   std_logic(7 downto 0)
  );
end entity NESTED_LOOP;

architecture RTL of NESTED_LOOP is

begin

  PROC : process (aclk) is

    procedure nested_loop_p is
    begin
      for j in 0 to 1 loop
        for i in 0 to 3 loop
          DATA(4 * j + i) := '1';
        end loop;
      end loop; -- this line's indentation is marked as invalid
    end procedure nested_loop_p;

  begin

    if (clk'event and clk = '1') then
      nested_loop_p;
    end if;

  end process PROC;

end architecture RTL;
~~~

Resulting output:

~~~
File:  \nested_loop.vhd                                      
==========================                                      
Phase 1... Reporting                                            
Phase 2... Reporting                                            
Phase 3... Reporting                                            
Phase 4... Reporting                                            
  for_loop_002              |         26 | Invalid indentation. 
  procedure_003             |         26 | Invalid indentation. 
Phase 5... Not executed                                         
Phase 6... Not executed                                         
Phase 7... Not executed                                         
==========================                                      
Total Rules Checked: 260                                        
Total Violations:    2                                          
~~~

jeremiah-c-leary, 2020-05-04:
Greetings @tlbeng,

I will take a look at this tonight.

jeremiah-c-leary, 2020-05-05:
Hey @tlbeng,
I just pushed version 1.6.0.  It looks like it is working.  If you could please verify I would appreciate it.

tlbeng, 2020-05-05:
Hi Jeremiah,

that was very quick, thank you. Unfortunately I now get other false positives in my codebase. Here's an updated example:

~~~vhdl
-- Nested Loop example

library ieee;
  use ieee.std_logic_1164.all;

entity NESTED_LOOP is
  port (
    CLK             : in    std_logic;
    DATA            : out   std_logic(7 downto 0)
  );
end entity NESTED_LOOP;

architecture RTL of NESTED_LOOP is

begin

  PROC : process (aclk) is

    procedure nested_loop_p is
    begin
      for j in 0 to 1 loop
        for i in 0 to 3 loop
          DATA(4 * j + i) <= '1';
        end loop;
      end loop; 

      if (DATA(0) = '1') then
        DATA(1) <= '0';
      end if; -- now this line's indentation is marked as invalid

    end procedure nested_loop_p;

  begin

    if (clk'event and clk = '1') then
      nested_loop_p;
    end if;

  end process PROC;

end architecture RTL;
~~~~

~~~~
File:  nested_loop.vhd
==========================
Phase 1... Reporting
Phase 2... Reporting
Phase 3... Reporting
Phase 4... Reporting
  if_014                    |         29 | Invalid indentation.
  procedure_003             |         29 | Invalid indentation.
Phase 5... Not executed
Phase 6... Not executed
Phase 7... Not executed
==========================
Total Rules Checked: 260
Total Violations:    2
~~~~

jeremiah-c-leary, 2020-05-05:
@tlbeng ... hmmm...I am assuming this is similar to the previous issue.  I will check it out.

tlbeng, 2020-05-05:
I had a look myself and it seems like the "end if" in line 29 is interpreted as an "end procedure" in `_classify_end_procedure_in_package_body`, and therefore the indentation level is decremented twice.

jeremiah-c-leary, 2020-05-06:
@tlbeng , yep that was the issue.  I had a check to see if the line was an `end if`:
```python
def _classify_end_procedure_in_package_body(dVars, oLine):
    if re.match('^\s*end', oLine.lineLower) and not oLine.isEndIfKeyword and \
       not oLine.isEndCaseKeyword and not oLine.isForLoopEnd and not oLine.isWhileLoopEnd and \
       not oLine.insideForLoop:
```

However, the classification of if statements occurs after the procedure in vhdlFile.py:

```python
51        for sLine in self.filecontent:
52            oLine = line.line(sLine.replace('\t', '  ').rstrip())
53            update.inside_attributes(dVars, self.lines[-1], oLine)

75            classify.procedure(dVars, oLine, oLinePrevious)

87            classify.if_statement(dVars, oLine)
```
So the line attribute `oLine.isForLoopEnd` is not set when checking for the end of the procedure.

I pushed version 1.6.1.

Now I am curious what else you are going to find.

tlbeng, 2020-05-06:
Great work. Thank you!

closed 2020-05-06 by tlbeng