jeremiah-c-leary/vhdl-style-guide #384: assertion with less-than-or-equal (<=) comparison

tlbeng, 2020-05-06:
**Environment**
Windows 10, vsg 1.6.1, python 3.7.7

**Describe the bug**
An assertion statement inside a process is interpreted as a signal assignment if it contains a less-than-or-equal (<=) comparison.

**To Reproduce**
run vsg on this vhdl
~~~vhdl

architecture RTL of ASSERT_LEQ is

begin

  -- this is ok
  assert 3 <= 7
    report "something";

  PROC : process (CLK) is
  begin

    -- this is interpreted as an assignment
    assert 3 <= 7
      report "something";

  end process PROC;

end architecture RTL;
~~~
output:
~~~
File:  assert_leq.vhd                                              
=====================                                              
Phase 1... Reporting                                               
Phase 2... Reporting                                               
Phase 3... Reporting                                               
Phase 4... Reporting                                               
  assert_001                |         14 | Invalid indentation.    
  sequential_001            |         14 | Invalid indentation.    
Phase 5... Not executed                                            
Phase 6... Not executed                                            
Phase 7... Not executed                                            
=====================                                              
Total Rules Checked: 260                                           
Total Violations:    2                                             
Steps to reproduce the behavior:
~~~

**Expected behavior**
no violation should be reported

jeremiah-c-leary, 2020-05-08:
@tlbeng I am on it.

jeremiah-c-leary, 2020-05-09:
@tlbeng This will be included in the 1.7.0 release.

jeremiah-c-leary, 2020-05-10:
@tlbeng, I just released 1.7.0.  If you could verify it addresses this issue and close this issue if it does.

Thanks,

--Jeremy

closed 2020-05-11 by tlbeng