jeremiah-c-leary/vhdl-style-guide #385: variable lists in nested procedures and functions are truncated

tlbeng, 2020-05-06:
**Environment**
Windows 10, vsg 1.6.1, python 3.7.7

**Describe the bug**
when multiple variables are declared in a single line in a procedure or function within a process, 
the variable list is truncated when calling `vsg -f filename.vhd --fix`.

**To Reproduce**
call `vsg -f filename.vhd --fix` on this vhdl
~~~vhdl

architecture RTL of VARIABLE_LIST is

    procedure test_p (x: integer) is
      variable a, b : integer; -- this line survives unharmed
    begin
    end procedure;

begin

  PROC : process (CLK) is

    procedure test_p (x: integer) is
      variable a, b : integer; -- this line is butchered
    begin
    end procedure;

    function test_f (x: integer) return integer is
      variable a, b : integer; -- this line is butchered
    begin
    end function;
  begin
  end process PROC;

end architecture RTL;
~~~
produces this "fixed" vhdl.
~~~vhdl

architecture RTL of VARIABLE_LIST is

  procedure test_p (x: integer) is
    variable a, b : integer; -- this line survives unharmed
  begin
  end procedure;

begin

  PROC : process (CLK) is

    procedure test_p (x: integer) is
      variable a, : integer; -- this line is butchered
    begin
    end procedure;

    function test_f (x: integer) return integer is
      variable a, : integer; -- this line is butchered

    begin

    end function;

  begin

  end process PROC;

end architecture RTL;
~~~

jeremiah-c-leary, 2020-05-08:
@tlbeng hmm....this one is a puzzler.

jeremiah-c-leary, 2020-05-08:
@tlbeng  I isolated the issue down to rule `process_031`.  I am still deciding how to fix the problem.  Meanwhile you can disable the rule using the following YAML configuration file:

```yaml
rule:
  process_031:
    disable: True
```
Use the `-c` option to use the configuration file.

jeremiah-c-leary, 2020-05-09:
@tlbeng This will be included in the 1.7.0 release.

jeremiah-c-leary, 2020-05-10:
@tlbeng, I just released 1.7.0.  If you could verify it addresses this issue and close this issue if it does.

Thanks,

--Jeremy

tlbeng, 2020-05-11:
@jeremiah-c-leary, unfortunately this issue is not fixed. The fix just stops alignment of variable declarations inside functions completely, but the issue is with tokenizing declarations of the form
variable a, b : integer and had nothing to do with the declaration being inside a function. Here's another sample:

~~~vhdl

architecture RTL of VARIABLE_LIST is

begin

  PROC : process (CLK) is

    variable a, b : integer; -- this line is butchered
    variable c  : integer;

  begin

  end process PROC;

end architecture RTL;
~~~

jeremiah-c-leary, 2020-05-12:
@tlbeng I was able to replicate the error.  I am on it.

jeremiah-c-leary, 2020-05-12:
@tlbeng I just released version 1.8.0.  If you could verify this version addressed the issue and close this issue if it does.

Thanks,

--Jeremy

tlbeng, 2020-05-12:
The reported issue is resolved, but the alignment of the variable declarations at ':' does not work unless there is a long variable - it seems the variable list is not considered for determining the ':' column.

~~~vhdl

architecture RTL of VARIABLE_LIST is

begin

  PROC1 : process (CLK) is

    -- alignment at ':' does work
    variable longname : integer;
    variable c : integer;
    variable a, b : integer;

  begin

  end process PROC1;

  PROC2 : process (CLK) is

    -- alignment at ':' does not work
    variable c : integer;
    variable a, b : integer;

  begin

  end process PROC2;

end architecture RTL;
~~~
Should I close this issue and open a separate one?

tlbeng, 2020-05-12:
Btw thanks for being so responsive and quick to address issues.

jeremiah-c-leary, 2020-05-12:
@tlbeng I think I got it this time.

You have to be careful with the rule `file_003`, which conflicts with this one.  `file_003` wants a single space after the `file` keyword.  This is configurable and an example is shown in the documentation.  These rules were updated to a more consistent methodology in the future 2.0.0 release.  Which seems to be taking too long to get out.

Anyway...I just release version 1.8.1.  If you could check it out and let me know if it fixes the issue, I would appreciate it.

Thanks,

--Jeremy

closed 2020-05-12 by tlbeng