jeremiah-c-leary/vhdl-style-guide #389: instantiation_011 also parses comment in same line

staerz, 2020-05-09:
**Environment**
vsg 1.6.1, fedora 30

**Describe the bug**
In a port mapping like
```
        action    => bcr_i, -- this is the actual BCR recovered from the TTC
```
instantiation_011 raises a violation although the offending text (upper case) is the comment (which is checked by another rule).

**Expected behavior**
I would not expect this violation to pop up in this case.

**Additional context**
Might be a parsing issue. Suggestion: Ignore anything after a comment ("--").

jeremiah-c-leary, 2020-05-09:
@staerz I was able to replicate the issue with the following code:

```vhdl
                                                                                              
architecture RTL of ENT is
begin
U_BLAH : BLAH
  port map (
      PORT1  => w_port1,
      ACTION => bcr_i, -- this is the actual BCR recovered from the TTC
      PORT2  => w_port2
  );
end architecture RTL;
```

I am looking into it.

jeremiah-c-leary, 2020-05-09:
@staerz instantiation_011 was assuming comments were removed.  I changed an extract routine to remove comments from the line before checking for case on the ports.

This will be included in the 1.7.0 release.

staerz, 2020-05-11:
Dear Jeremy,

thanks, that fixes the problem!

closed 2020-05-11 by staerz