jeremiah-c-leary/vhdl-style-guide #390: entity_008 and entity_012 don't allow numbers

staerz, 2020-05-09:
**Environment**
VSG 1.6.1., Fedora 30

**Describe the bug**
entity_008 and entity_012 don't allow numbers in the entity name (tested with requiring lower case).

**To Reproduce**
Try on the following entity name:
```
entity s10_devkit_top is
```
with the following setting
```
    "entity_008":{
      "case": "lower"
    },
    "entity_012":{
      "case": "lower"
    },
```
**Expected behavior**
As numbers are valid (except at the start of the entity name), this must be allowed and not be marked as a violation.

The issue might likely show up in other case sensitive rules (not tested).

jeremiah-c-leary, 2020-05-09:
@staerz I am looking at this now.

jeremiah-c-leary, 2020-05-09:
@staerz  I was unable to duplicate the issue with the following file:

```vhdl

entity s10_devkit_top is
  port (
    A : in    std_logic;
    B : out   std_logic
  );
end entity s10_devkit_top;
```

and the following configuration:

```yaml
rule:
  entity_008:
    case: lower
  entity_012:
    case: lower
```

Is there additional information you can give me to help isolate the issue?

jeremiah-c-leary, 2020-05-10:
@staerz , I just released 1.7.0.  If you could verify it addresses this issue and close this issue if it does.

Thanks,

--Jeremy

staerz, 2020-05-11:
Dear Jeremy,

with VSG 1.7.0 I don't get the problem anymore. I do agree, it is a bit awkward that you couldn't reproduce the problem, but maybe there was some underlying function(ality) that was changed so now the problem doesn't show up anymore.

Thanks for this very rapid fix!

closed 2020-05-11 by staerz