jeremiah-c-leary/vhdl-style-guide #391: instantiation_030 raises in case generate (VHDL-2008)

staerz, 2020-05-09:
**Environment**
VSG 1.6.1, Fedora

**Describe the bug**
instantiation_030 raises in case generate (VHDL-2008)

**To Reproduce**
Try on the following piece of code:
```
  gen_delay : case d_depth generate
    --! No delay of #sig_i
    when 0 => 
      delay_reg(0) <= sig_i;
    when 1 =>
      --! Delay #sig_i by one clock cycle
      proc_singleDelay : process (clk)
      begin
        if rising_edge(clk) then
          delay_reg(0) <= sig_i;
        end if;
      end process;
    when others =>
      --! Delay #sig_i by multiple clock cycles
      proc_multipleDelay : process (clk)
      begin
        if rising_edge(clk) then
          --! shift right and insert sig_in on the left
          delay_reg(d_depth-1 downto 0) <= sig_i & delay_reg(d_depth-1 downto 1);
        end if;
      end process;
  end generate;
```

Note that the "case .... generate" is VHDL-2008.

**Expected behavior**
I would not expect instantiation_030 to raise a violation here.

jeremiah-c-leary, 2020-05-09:
@staerz Agree that this is an issue.  I would appear the generate is being seen as an instantiation.  I am looking into it.

jeremiah-c-leary, 2020-05-10:
@staerz An update for this issue will be in the 1.7.0 release.

jeremiah-c-leary, 2020-05-10:
@staerz , I just released 1.7.0.  If you could verify it addresses this issue and close this issue if it does.

Thanks,

--Jeremy

staerz, 2020-05-11:
Hi Jeremy,

yes, I confirm, VSG 1.7.0. doesn't have this issue anymore!

Thanks for being so quick in addressing it, really appreciated!

closed 2020-05-11 by staerz