jeremiah-c-leary/vhdl-style-guide #392: Case ... generate not properly recognised in indentation (VHDL-2008)

staerz, 2020-05-09:
**Environment**
VSG 1.6.1., Fedora

**Describe the bug**
Running on the following code (with mostly default settings)
```
  --! now set the last item in the chain
  gen_delay : case d_depth generate
    --! No delay of #sig_i
    when 0 => 
      delay_reg(0) <= sig_i;
    --! Delay #sig_i by one clock cycle
    when 1 =>

      proc_singleDelay : process (clk)
      begin
        if rising_edge(clk) then
          delay_reg(0) <= sig_i;
        end if;
      end process;

    --! Delay #sig_i by multiple clock cycles
    when others =>

      proc_multipleDelay : process (clk)
      begin
        if rising_edge(clk) then
          --! shift right and insert sig_in on the left
          delay_reg(d_depth-1 downto 0) <= sig_i & delay_reg(d_depth-1 downto 1);
        end if;
      end process;

  end generate;
```
end up in the following list of violations:
```
======================================================
Phase 1... Reporting
Phase 2... Reporting
Phase 3... Reporting
Phase 4... Reporting
  comment_005               |         83 | Align comment with "when" keyword.
  concurrent_001            |         85 | Invalid indentation.
  comment_005               |         86 | Align comment with "when" keyword.
  process_001               |         89 | Invalid indentation.
  process_003               |         90 | Invalid indentation.
  if_001                    |         91 | Invalid indentation.
  sequential_001            |         92 | Invalid indentation.
  if_014                    |         93 | Invalid indentation.
  process_006               |         94 | Invalid indentation.
  comment_005               |         96 | Align comment with "when" keyword.
  process_001               |         99 | Invalid indentation.
  process_003               |        100 | Invalid indentation.
  if_001                    |        101 | Invalid indentation.
  sequential_001            |        103 | Invalid indentation.
  if_014                    |        104 | Invalid indentation.
  process_006               |        105 | Invalid indentation.
Phase 5... Not executed
Phase 6... Not executed
Phase 7... Not executed
```

It seems that the VHDL-2008 construct "case ... generate" is not properly taken into account in the indentation checks.

**Expected behavior**
I would not expect any violation of indentation for the code given above.

Please let me know if I'm overlooking anything.

jeremiah-c-leary, 2020-05-10:
@staerz I did not have support for case generates so I am not surpised you received invalid warnings.  I believe I have made the necessary changes.  The updates will be in the 1.7.0 release.

jeremiah-c-leary, 2020-05-10:
@staerz , I just released 1.7.0.  If you could verify it addresses this issue and close this issue if it does.

Thanks,

--Jeremy

staerz, 2020-05-11:
Hi Jeremy,

thanks, yes, that solves the problem! :+1:

closed 2020-05-11 by staerz