jeremiah-c-leary/vhdl-style-guide #395: Release Notes

staerz, 2020-05-12:
**What is your question?**
Dear Jeremiah,

thank you very much for putting so much effort into improving this tool. That's very appreciated!

I would have the question if you have release notes for each version. You release quite often, which is very good, but it is (as far as I can see) not easy to see what the new features or bug fixes are.

But I'm not keen on github, so if you could just point me to a link, that would be helpful. A minimum would be a list of issues that have been treated. I have a one-liner for this if you are interested.

jeremiah-c-leary, 2020-05-12:
Thank you @staerz , I appreciate the compliment.

For release information you can check this link:  https://github.com/jeremiah-c-leary/vhdl-style-guide/releases

[image]

If you click on the ellipses next to the release number:
[image]

it will show you which issues were addressed.  The issue number in the release is also a link so you can go to the issue itself.  That can be really handy.

staerz, 2020-05-12:
Haha, yes, I knew it would be somewhere.
I found the link thanks to your help!

closed 2020-05-12 by staerz