jeremiah-c-leary/vhdl-style-guide #396: configuration option allow_single_line (concurrent_007) and clock (process_029)

staerz, 2020-05-12:
**What is your question?**
Dear Jeremiah,

I'm having difficulties in setting attributes for certain rules.

I try setting this in a configuration file that I load via the -c option: disabling works, but not setting the intended attribute :confused: 
```python
{
  "file_list":[
    "modules/fpga/src/*/*.vhd",
    "modules/fpga/src/*.vhd",
  ],
  "local_rules":".vsg/rules",
  "rule":{
    "global":{
      "indentSize":2
    },
    # somehow I can't get this to work...
    "concurrent_007":{
      "disable":true
#        "allow_single_line":true
    },
    # somehow I can't get this to work...
    "process_029":{
      "disable":true
#      "clock":"edge"
    },
  },
}
```
Would you see an obvious mistake?

It would help if the explanation to the rules would have an explicit configuration example.

Configuring the default upper or lower case though works fine:
```
  "rule":{
    "architecture_013":{
      "case": "lower"
    },
    ...
```

From https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html: is it possible that this is incomplete or contains a mistake:
```python
{
  "file_list":[
    "entity.vhd":{
      "rule":{
        "length_001":{
          "disable": true
        }
      }
   },
   "architecture.vhd",
   "package.vhd"
  ],
  "rule":{
    "global":{
      "disable": true
    },
    "rule": {  ## this looks strange to me - that syntax fails in my trials
      "length_001":{
        "disable": false
    }
  }
}
```

jeremiah-c-leary, 2020-05-13:
@staerz, you caught us before the 2.0.0 release.  We are making some architectural changes to how the rules work and improving the user experience. 

At the moment I have a release_1.0_support branch I am cherry-picking fixes to, from master, until 2.0.0 is ready.  The `latest` version is from master.  I did enable building from the `release_1.0_support` branch.

If you look at the bottom left of the read the docs webpage you will see the following:

[image]

If you click on the v:latest you will get the option to pick a different version:
[image]

This is a long winded way of saying the options you want to set for those rules do not exist in the `release_1.0_support` branch.  Due to the changes made to the rule structure migrating them over is difficult.

You might want to consider switching to the development version.  Take a look at the 2.0.0 project kanban board https://github.com/jeremiah-c-leary/vhdl-style-guide/projects/22.  This will give you access to those parameters and a lot of other new features.

Most of the changes are under the hood except for issue #343, for which I want issue #350 completed first.

If you follow the Git Hub installation instructions, you can install the latest version.

[image]

staerz, 2020-05-13:
Dear @jeremiah-c-leary,

thank you very much for this exhaustive response.

That explains things pretty well and I see you're also very well organised!

No need to rush. I'll stick to the official releases as we use VSG with multiple developers and even in our pipelines (that you added the JUnit support was one reason to go for VSG - that's just the best!).

I will keep reporting issues if we stumble over some, but will always check if an issue is already reported.
If you say that "the options you want to set for those rules" are not yet supported, I'm perfectly fine with that since I now know that you have it on your todo list.

Thanks and I'll stay tuned!

jeremiah-c-leary, 2020-05-16:
@staerz ...as I was implementing issue #350 I was reminded of something you had mentioned:

> It would help if the explanation to the rules would have an explicit  configuration example.

Have you tried the `-rc` option?

I am currently going through defining the style I follow so I have to configure rules differently.  I use this option to see how a rule can be configured.

```mono
$ vsg -rc entity_017
{
  "rule": {
    "entity_017": {
      "indentSize": 2,
      "phase": 5,
      "disable": false,
      "fixable": true,
      "compact_alignment": true,
      "blank_line_ends_group": true,
      "comment_line_ends_group": true,
      "separate_generic_port_alignment": true
    }
  }
}
```
I then add the rule configuration to my overall configuration and change what I need.  I usually drop the indentSize, phase, disable and fixable items.

```yaml
# The name must be a single word
name : legacy
# Define the rules for this style
rule :
  entity_008 :
    case : upper
  entity_012 :
    case : upper
  port_010 : 
    case : upper
  entity_017 :                                                                                    
    compact_alignment : false                                                                     
    blank_line_ends_group : false                                                                 
    comment_line_ends_group : false
```

staerz, 2020-05-18:
Dear @jeremiah-c-leary,

thanks! Good point!

Here is what I get:
```
    "process_029": {
      "disable": true,
      "fixable": true,
      "indentSize": 2,
      "phase": 1
    },
    "concurrent_007": {
      "disable": true,
      "fixable": true,
      "indentSize": 2,
      "phase": 1
    },
```
for the 2 rules in question.

So as you state, the other options are not yet available there... I'll hence be patient. No worries and thanks for pointing to the option again.

jeremiah-c-leary, 2020-07-18:
Hey @staerz ,

I just released version 2.0.0.  Could you double check that it addresses the issues you identified in this ticket.  If so, could you close this ticket?

Thanks,

--Jeremy

staerz, 2020-07-20:
Hi @jeremiah-c-leary,

yes, VSG 2.0.0 fixes this issue, hence closing :+1:

closed 2020-07-20 by staerz