jeremiah-c-leary/vhdl-style-guide #398: Fix introduces syntax error on generics that start with "port"

ericjmcd, 2020-05-18:
**Environment**
Ubuntu 18.04
VSG version 1.8.1

**Describe the bug**
If a generic name starts with "port" (or "PORT") the fixer assumes that is the beginning of the "port" keyword and mangles the output.

**To Reproduce**
Example original:
```
entity EXAMPLE is
  generic (
    PORT_CNT : integer := 1);
    port (
    CLK     : in    std_logic;
  );
end entity EXAMPLE;
architecture BEHAVIOR of EXAMPLE is
begin
end architecture BEHAVIOR;
```

Example after above is "fixed":
```entity EXAMPLE is
  generic (
    port (_CNT : integer := 1
  );
    port (
    CLK     : in    std_logic;
  );
end entity EXAMPLE;

architecture BEHAVIOR of EXAMPLE is

begin

end architecture BEHAVIOR;
```

jeremiah-c-leary, 2020-05-18:
@ericjmcd I think I know what is going on.  I suspect the `PORT_CNT` is being classified as the start of the port declaration.  I will try to have a fix tomorrow.

jeremiah-c-leary, 2020-05-19:
@ericjmcd I just released version 1.9.0.  Could you verify it fixes the identified issue, and it so could you close this issue?

Thanks,

--Jeremy

ericjmcd, 2020-05-19:
@jeremiah-c-leary Thanks for the quick bugfix!  1.9.0 looks good.  No issues with generics with names: "PORT_CNT" and even "PORT" or "port".

closed 2020-05-19 by ericjmcd