jeremiah-c-leary/vhdl-style-guide #400: Rule process_029 documentation is wrong for clock events vs edges.

ericjmcd, 2020-05-21:
**Environment**
Version 1.9.0

**Describe the bug**
Docs call for setting "clock": "edge" or "event" but now process_029 is always for events so need to disable it to allow "edge" setting.  I actually preferred the selection (i.e. all rising_edge or all event).  This now limits it to all event or a mix of edges/events.

abyszuk, 2020-05-22:
Can we get back the selection feature? I will really miss that.

jeremiah-c-leary, 2020-05-22:
Hey @ericjmcd That option will be available in the 2.0.0 release.  The `latest` version of the documentation on readthedocs is pointing to the HEAD of `master`.  The 1.x releases are being maintained on the `release_1.0_support` branch.

You can find the 1.9.0 documentation by finding the `v:latest` button on the bottom left of the window:
[image]

If you click it, then you will get a list of versions:
[image]

So you can pick the one that matches your installed version.

I really should make the `release_1.0_support` branch the latest until I get 2.0.0 out.

jeremiah-c-leary, 2020-05-22:
Hey @abyszuk ,
I am not sure it was in any of the 1.x releases.  It came in on issue #319 which was scheduled for the 2.0.0 release.

ericjmcd, 2020-06-01:
@jeremiah-c-leary Thanks for pointing out the 2.0.0 release with the fix and my wrong docs versions.  Considering this closed as it will be resolved in 2.0.0.

closed 2020-06-01 by ericjmcd