jeremiah-c-leary/vhdl-style-guide #401: All phase errors

qarlosalberto, 2020-05-30:
Hello,

I'm adding VSG support to TerosHDL vscode extension (https://github.com/TerosTechnology/vscode-terosHDL).  Is it possible to pass all phases? Or do I need to fix the previous phase errors?
I would like to show all errors at the same time, so the user can select what to fix.

jeremiah-c-leary, 2020-06-01:
Hey @qarlosalberto,

I had not originally thought to run through all the phases.  The original concept was to first everything in one phase before moving on to others.  The thinking being that later phases would be dependent on earlier ones.

I suppose there is no reason why that has to be enforced though.  I could easily add an option to not stop on errors.

It all depends on how you envision incorporating VSG into terosHDL.  Would it just be reporting issues?

I am looking forward to further discussions on this.

Regards,

--Jeremy

qarlosalberto, 2020-06-02:
Thanks for the reply @jeremiah-c-leary 

I have an initial version of TerosHDL with VSG support:

[image]

My idea (although it may not be in your development plans) is to have a similar feature as others style linters.

1. Show all style erros.
2. Ability to fix a specific error.
3. Comments to skip a line error.

What would be the operation mode?

1. TerosHDL runs VSG and get all style errors (it would be easier if it's possible to configure the output as json)
2. TerosHDL user requests to fix a problem.
3. TerosHDL requests to VSG to fix the problem (line + problem code).
4. VSG fixs the problem and return the code with this error fixed.

Here you can see an example with other style linter (for typescript):

[image]

[image]

[image]

jeremiah-c-leary, 2020-06-02:
> I have an initial version of TerosHDL with VSG support:

That is amazing.  Is that in Visual Studio Code?  At one time I was trying to figure it out, but I gave up after other pressing concerns.  I do not use VSG that way, but I can see the appeal.

What is performance of running VSG from VSC?  I use vim, and there is a noticeable delay.

> My idea (although it may not be in your development plans) is to have a similar feature as others style linters.

>Show all style erros.

I think could be done.  We would have to watch out for new errors as others are fixed.
Alignment errors can be tricky.  If there are several applied to a single line, the order would matter.  I suppose the worst affect would be you fix one, then fix the other, then you have to go back and fix the first one again.

That brings up a question:  How are multiple errors per line handled in Visual Studio Code?

>Ability to fix a specific error.

VSG currently does not support fixing a single violation.  It doesn't fit the current use model.  It could be added though.  There would have to be some way to pass the violation and the line number to it.  Maybe add an attribute to the base rule class which would limit the fix to the line numbers listed.

>Comments to skip a line error.

VSG has code tags, but they were envisioned as brackets around code.  The `disable-next-line` could be implemented as an alternative.

>What would be the operation mode?

>TerosHDL runs VSG and get all style errors (it would be easier if it's possible to configure the output as json)

JSON output could be added.

Maybe there should be a version of VSG dedicated for editors?  The underlying engine would be the same, but the command line interface could be tuned.  I could make `--fix` take an optional JSON file which could limit what is fixed.

qarlosalberto, 2020-06-04:
I'm glad you consider the proposals :)

Yes, it's Visual Studio Code. I have written the VSG linter in a independent library (https://github.com/TerosTechnology/colibri) and use it in the VSC extension. So it can be used in others editors (Atom, vim... etc...). It has a good performance but, if I'm honest, I only lint "on save" and only one file. But it can be extended to lint the full workspace and "on change".

I call to VSG, get .xml output and parse it. Later I send to VScode an array of errors with: description and line. So if there are multiple errors in the same line it will show it all. The pop-up will have 1,2,3... lines, one per error.

Currently I have a limited time to develop the extension. In the first version I can only add the linter errors display, the option to fix the problem with "disable-next-line" and fix all problems.
And in the next release I can add the support to fix a single violation.

It's a really good idea the option to pass a JSON!

jeremiah-c-leary, 2020-06-06:
@qarlosalberto so it sounds like there are several changes required to support your effort.  Does this list summarize what you need?

1. JSON formatted output indicating rules that failed with the line numbers and the solution.
2. Ability to disable a rule on the next line
3.  Show all style errors (Do not stop if a violation was found)
4.  Ability to pass a JSON file to tell VSG which errors to fix

If so, could you prioritize this list?

qarlosalberto, 2020-06-06:
That list is really nice! 
The priorized list could be:

2. JSON formatted output indicating rules that failed with the line numbers and the solution.
3. Ability to disable a rule on the next line
1. Show all style errors (Do not stop if a violation was found)
4. Ability to pass a JSON file to tell VSG which errors to fix

jeremiah-c-leary, 2020-06-07:
awesome,

I was going to create a ticket for each item in the list.  It will be easier to track them and you will know when each is ready.

jeremiah-c-leary, 2020-06-07:
Do you want the JSON output to be sent to the screen or to a file?

qarlosalberto, 2020-06-07:
Better to file

jeremiah-c-leary, 2020-06-08:
I have issue #416 implemented.  How does this look for a JSON output:

```json
{
  "violations": [
    {
      "rule": "concurrent_005",
      "linenumber": 580,
      "solution": "Remove label on concurrent assignment."
    },
    {
      "rule": "concurrent_005",
      "linenumber": 581,
      "solution": "Remove label on concurrent assignment."
    },
    {
      "rule": "concurrent_005",
      "linenumber": 582,
      "solution": "Remove label on concurrent assignment."
    },
```
It is a list of violations.  I used the `violations` key in case I need to pass other information that is not a violation.  Trying to future proof this a little.

jeremiah-c-leary, 2020-06-08:
Is the goal to use the next line tags sparingly?  I could see a case where your code could be cluttered with VSG code tags.

jeremiah-c-leary, 2020-06-08:
Hey @qarlosalberto ,

I have the first three items implemented on the issue-401 branch.  I think I will have to postpone the forth one for a bit while I work on some other tickets for 2.0.0

Could you pull down that branch and play with the updates?  Keep in mind that you will be pulling down a pre 2.0.0 version.

Thanks,

--Jeremy

qarlosalberto, 2020-06-08:
@jeremiah-c-leary no problem. It's really nice to have that features :)

jeremiah-c-leary, 2020-06-20:
@qarlosalberto 
My apologies, work kind of blew up on me.  I wanted to check if you were waiting on me for anything.

Thanks,

--Jeremy

qarlosalberto, 2020-06-23:
Thak you @jeremiah-c-leary !

The only one is this: https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/419   Or do you prefer to add it in the next version?

jeremiah-c-leary, 2020-12-06:
Hey @qarlosalberto ,

Just wanted to give you an update.  I am finally is a position to start transitioning this to the master branch for the next release.

Regards,

--Jeremy

jeremiah-c-leary, 2020-12-06:
lol....this branch is really out of date:

[image]

jeremiah-c-leary, 2020-12-06:
I merged the all phases and json command line arguments over.

I will start working on the single line exclusions and issue #419

--Jeremy

qarlosalberto, 2020-12-09:
Thank you!! @jeremiah-c-leary

jeremiah-c-leary, 2021-02-13:
Hey @qarlosalberto ,

Just wanted to ping you about the three issues I have for you and check if you have had a chance to give them a try.

Thanks,

--Jeremy

qarlosalberto, 2021-02-19:
Hey @jeremiah-c-leary thank you! I will try to check it the next week :)

jeremiah-c-leary, 2021-03-16:
Hey @qarlosalberto ,

I am going to close this issue for release 3.0.0 similar to the other issue I just closed.  If there are any updates regarding this we can open a new issue.

--Jeremy

closed 2021-03-16 by jeremiah-c-leary

qarlosalberto, 2021-03-25:
Thank you @jeremiah-c-leary :)