jeremiah-c-leary/vhdl-style-guide #402: Warning or Severity option for rules

ericjmcd, 2020-06-01:
I would like to have a feature where we could flag a rule with warnings only.  i.e. If the rule is violated, a warning message (or violation message) is still printed but the return code is still zero.  I'm using VSG with Jenkins and pre-commit hooks so I would like to allow some rule violations to occur with warnings without causing the check/build to fail.

Ideally, I'd like there to be a severity that is configurable per rule so that users could tailor the importance of each.  Maybe something similar to: violation, warning, format.

I'm trying to get my team to adopt VSG and I don't want to force them to drastically modify their code to pass or to disable a bunch of rules that we actually want to enforce.

jeremiah-c-leary, 2020-06-02:
Hey @ericjmcd 
I think such a thing could be done.  There are some implications to be worked through.

I am curious about this statement:
> I'm trying to get my team to adopt VSG and I don't want to force them to drastically modify their code to pass or to disable a bunch of rules that we actually want to enforce.

I would assume your team would use the `--fix` option to bring their code into compliance.

Could you elaborate on this a little?

ericjmcd, 2020-06-02:
You're right - I forgot that the --fix option was there (although I added it for my "make reformat" target).  I was thinking it was unreliable due to to the "generic name with 'port'" bug that I submitted (and you quickly fixed - thanks!) but I haven't had any problems since.  Once we get our config file to match the style that we like, I think we'll apply --fix liberally to get our projects in line.

PS - Thanks for making such a great tool.  If I wasn't already overcomitted to too many other things, I'd love to contribute.  Maybe someday I will.

jeremiah-c-leary, 2020-06-07:
> PS - Thanks for making such a great tool. If I wasn't already overcomitted to too many other things, I'd love to contribute. Maybe someday I will.

Thanks for the kind words, and I am glad you are finding it useful.

How soon would you like to see this implemented?  I am trying to figure out priorities between several tickets.

ericjmcd, 2020-06-08:
It's not a major hurry for us - just something that will make the tool more useful so we can soften the pass/fail criteria for our CI/CD pipeline (I could pipe the output to file and parse it and then report the warnings - I wish jUnit supported warnings but I don't believe it does).
For implementation, what about:
* add a global "severity" with either "error" or "warning" (defaulting to "error")
* allow rules to change the global "severity" to "error" or "warning"
* if a rule fails with "warning" severity, don't record it as a failure but still output it to stdout/sterr but change the output:
- * from "ERROR" to "WARNING" for syntastic output format?
- * prepend a WARNING prefix to the rule violation output for the VSG output format?

I'd prefer the errors/warnings to go to stderr instead of stdout but I'm just getting greedy :-)

jeremiah-c-leary, 2020-06-09:
> (I could pipe the output to file and parse it and then report the warnings - I wish jUnit supported warnings but I don't believe it does)

Take a look at issue #417.  If I added sections for errors and warnings etc... you could just read the JSON file in.

> add a global "severity" with either "error" or "warning" (defaulting to "error")

Agree.

> allow rules to change the global "severity" to "error" or "warning"

I was thinking of reserving the error/violations and letting the user decide how to categorize the rest.  Maybe somebody wants different categories:  'Fix_later', 'update_before_design_review'.  That might be too much flexibility.

>if a rule fails with "warning" severity, don't record it as a failure but still output it to stdout/sterr but change the output:

The output will have to change to indicate the different severity levels.  They could be grouped by severity.  We will have to play around with it and see what looks good.

> I'd prefer the errors/warnings to go to stderr instead of stdout but I'm just getting greedy :-)

Would the JSON format address this?

jeremiah-c-leary, 2020-07-04:
Hey @ericjmcd and @nfrancque 

I have this feature mostly implemented, but I ran into something I would like some feedback on.  There are two conditions to consider.

First condition, running VSG without the `--fix` option:

* If a rule is configured as a warning, then use the rule against the code but do not stop if there is a failure.
* If a rule is configured as an error, then use the rule against the code and stop if there is a failure 

Second condition, running VSG with the `--fix` option:

* If a rule is configured as an erro, then fix any violations detected
* If a rule is configured as a warning, then do not fix a violation if one is detected?

I am not sure what the proper response to a warning should be when the `--fix` option is used.

Opinions?

ericjmcd, 2020-07-04:
I think the whole purpose of the warning is to allow for things that you don't generally want so fixing them would defeat the purpose.  For example, I sometimes put comments on lines in the port map of an instantiation - usually TODOs. I wouldn't want those fixed.

I think we'll use warnings as "soft rules/recommendations" and let the developers decide for themselves although they'll be tracked in Jenkins and maybe even fail the build if there are too many.

jeremiah-c-leary, 2020-07-04:
So it sounds like we should only fix errors.  I think that makes sense.

Do you want to include warnings in the JUnit XML file?

Here is an example output with severities:

```mono
================================================================================
File:  debug/entity.vhd
================================================================================
Phase 6 of 7... Reporting
Total Rules Checked: 387
Total Violations:    4
Error : 0
Warning : 3
Guideline : 1
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  entity_019                | Warning    |          5 | Add the entity name.
  whitespace_012            | Guideline  |          1 | Remove all but 1 blank lines below.
  entity_016                | Warning    |          5 | Remove blank line(s) above "end entity" keywords.
  entity_008                | Warning    |          3 | Change entity name to lowercase.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

The built in severities are `Error` and `Warning`.  I am allowing the user to define their own severities.

```json

{
  "severity":{
    "Guideline":{
      "type":"warning"
    }
  },
  "rule":{
    "entity_016":{
      "severity":"Warning"
    },
    "entity_019":{
      "severity":"Warning"
    },
    "whitespace_012":{
      "severity":"Guideline"
    }
  }
}
```
...as long as they use an existing type of severity.

jeremiah-c-leary, 2020-07-04:
Hey @ericjmcd ,

I just created a pull request, #439, for this issue.  Could you take a look at it and pull down the branch and kick the tires a little.

Thanks,

--Jeremy

ericjmcd, 2020-07-05:
@jeremiah-c-leary that looks fantastic.  Unfortunately (or I guess "fortunately"), I'm going on a vacation tomorrow for a week so I won't be able to try it out until I get back.

jeremiah-c-leary, 2020-07-05:
@ericjmcd ,
Have a good vacation.

closed 2020-07-28 by jeremiah-c-leary