jeremiah-c-leary/vhdl-style-guide #404: signal_015 false positive when function used to constrain size

joshrsmith, 2020-06-02:
**Environment**
```
$ pip list
Package    Version
---------- -------
pip        20.1.1
PyYAML     5.3.1
setuptools 47.1.1
vsg        1.9.0
wheel      0.34.2
```

**Describe the bug**
VSG flags a false error for a line of code that looks similar to this, when signal_015 rule is enabled:
`signal foo : std_logic_vector(maximum(G_A, G_B) + maximum(C_A, C_B)-1 downto 0);`

The error is: `multiple signals defined in a single declaration`

When trying to use `--fix` the following stack trace results:

```
  File "vhdl-style-guide/bin/vsg", line 8, in <module>
    sys.exit(main())
  File "vhdl-style-guide/lib/python3.7/site-packages/vsg/__main__.py", line 180, in main
    oRules.fix(commandLineArguments.fix_phase)
  File "vhdl-style-guide/lib/python3.7/site-packages/vsg/rule_list.py", line 136, in fix
    oRule.fix(self.oVhdlFile)
  File "vhdl-style-guide/lib/python3.7/site-packages/vsg/rule.py", line 71, in fix
    self._fix_violations(oFile)
  File "vhdl-style-guide/lib/python3.7/site-packages/vsg/rules/signal/rule_015.py", line 49, in _fix_violations
    oLine.update_line('  signal ' + lSignals[i] + ' : ' + sAfterColon)
IndexError: list index out of range
```

**To Reproduce**
See above test case.

If `--fix` is not used, then the exception is not thrown, but the false positive is still present.

**Expected behavior**
No error should be reported, no `--fix` should be necessary.

**Screenshots**
None

**Additional context**
I am guessing that the comma in the function call is causing the problem?

jeremiah-c-leary, 2020-06-03:
Hey @joshrsmith ,
Thanks for the bug report.  I will take a look at it.

jeremiah-c-leary, 2020-06-04:
Hey @joshrsmith ,

I just released version 1.10.0 which has a fix for this issue.  Could you please update and verify it addresses the issue for you.

Thanks,

--Jeremy

joshrsmith, 2020-06-04:
Fixed.

closed 2020-06-04 by joshrsmith