jeremiah-c-leary/vhdl-style-guide #405: `--fix` for process_018 does not re-use capitalization of beginning label

joshrsmith, 2020-06-02:
**Environment**
python 3.7.2, Linux

```
$ pip list
Package    Version
---------- -------
pip        20.1.1
PyYAML     5.3.1
setuptools 47.1.1
vsg        1.9.0
wheel      0.34.2
```

**Describe the bug**
If a process has been defined as follows:
```
My_Process : process(clk)
   ...
end process;
```
and `process_018` is enabled, and the `--fix` option is specified, then the resulting fix will be:
```
My_Process : process(clk)
   ...
end process my_process;
```
Even if `process_017` and `process_019` are both disabled.

**To Reproduce**
See above.

**Expected behavior**
I would expect that capitalization would be preserved from the original label if case is not being enforced on the label via `process_017` and `process_019`. Therefore I would expect the fix to be:
```
My_Process : process(clk)
   ...
end process My_Process;
```
If `process_019` were enabled, I would expect the capitalization to follow that configuration.

**Screenshots**
None.

**Additional context**
None.

jeremiah-c-leary, 2020-06-04:
Hey @joshrsmith ,
I agree the case should be preserved if `process_017` and `process_019` are disabled.

I am working on this.

jeremiah-c-leary, 2020-06-04:
Found the issue.  There was a regular expression match being done on the lower cased version of the line.  This resulted in the lower case version of the label being used.

jeremiah-c-leary, 2020-06-04:
Hey @joshrsmith ,

I just released version 1.10.0 which has a fix for this issue.  Could you please update and verify it addresses the issue for you.

Thanks,

--Jeremy

joshrsmith, 2020-06-04:
Yes, I can confirm this has been resolved.

closed 2020-06-04 by joshrsmith