jeremiah-c-leary/vhdl-style-guide #406: port_013 ("Place multiple ports on their own lines") false positive

joshrsmith, 2020-06-04:
**Environment**
```
vsg               1.10.0
Python 3.7.2, 64-bit, Linux
```

**Describe the bug**
port_013 triggers a false positive when a port looks like the following:
`I_INPUT : in T_CUSTOM_ARRAY(G_A'high downto 0)(function_call(G_B,G_C)-1 downto 0) := (others => (others => '0'));`
(For clarity, this is all on one line without any linebreak in the middle)

The false positive is not triggered if the line looks like this:
`I_INPUT : in T_CUSTOM_ARRAY(G_A'high downto 0)(function_call(G_B,G_C)-1 downto 0);`

It appears to have something to do with the initial value assignment.

**To Reproduce**
See above

**Expected behavior**
In both cases above, there is a single port declared on the line. No error should be flagged in either case.

**Screenshots**
N/A

**Additional context**
N/A

jeremiah-c-leary, 2020-06-06:
@joshrsmith I am starting to look at this.

jeremiah-c-leary, 2020-06-06:
Hey @joshrsmith 

I just released version 1.11.0.  Could you please validate it resolves this issue.

Thanks,

--Jeremy

joshrsmith, 2020-06-08:
Resolved.

closed 2020-06-08 by joshrsmith