jeremiah-c-leary/vhdl-style-guide #407: instantiation_024 (Rewrite positional assignment to explicit assignment) false positive

joshrsmith, 2020-06-04:
**Environment**
```
vsg               1.10.0
Python 3.7.2, 64-bit, Linux
```

**Describe the bug**
If an instance is coded like this, then a false positive for instantiation_024 is generated:

```
My_Unit : entity work.MY_ENTITY
generic map (
    G_A => C_A(i),  -- OK
    G_B => C_B,     -- OK
    G_C => C_C(i),  -- OK
    G_D => C_D(i),  -- OK
    G_E => resize(G_E1(i), C_E2(i), G_E3), -- triggered
    G_F => resize(G_F1(i), C_F2(i), G_F3)  -- triggered
)
port map(
    ...
```
I suspect it is related to using a function as the actual for the generic in the map.

Note that this unit is generated in a generate loop (hence the index `i`), but I am not sure if that is related.

**To Reproduce**
See above.

**Expected behavior**
I would expect the error not to be triggered, because explicit assignment is being used.

**Screenshots**
N/A

**Additional context**
N/A

jeremiah-c-leary, 2020-06-06:
Hey @joshrsmith I am starting to look at this one.

jeremiah-c-leary, 2020-06-06:
Hey @joshrsmith 

I just released version 1.11.0.  Could you please validate it resolves this issue.

Thanks,

--Jeremy

joshrsmith, 2020-06-08:
Resolved.

closed 2020-06-08 by joshrsmith