jeremiah-c-leary/vhdl-style-guide #408: New rule - Maximum number of lines in file

nfrancque, 2020-06-04:
Perhaps in conjunction with #402, it would be nice to have a warning that a file is too long with some suggestions as to how to break it up.  Suggestions could in the future be improved by metrics about how many signals, processes, concurrent statements, etc exist in the file against some known average.  This is more advanced than necessary but could be nice down the line.  

There are obvious justifications for breaking this rule, so I would definitely make it a warning if #402 is implemented.

jeremiah-c-leary, 2020-06-07:
Could there be different levels for the same rule?  For example, the file length.  You get a warning when it hits 1000 lines, but a violation at 3000 lines?

nfrancque, 2020-06-07:
I think that is reasonable.  Not sure how easily it applies to other warnings though, I think these count-based rules are the only ones that fall under that category.

jeremiah-c-leary, 2020-07-18:
what should the default value be, and I would make this a warning versus an error.

nfrancque, 2020-07-19:
Agree, if we can't do multiple levels by the count would make it a warning.  How does 500 sound?

jeremiah-c-leary, 2020-12-08:
Hey @nfrancque ,

I just implemented this and made the default 2000 lines.  That should catch excessively long files.  It is defaulted to be being a warning.

This will be in the 3.0.0 release.

Hope things are going well for you.

--Jeremy

closed 2020-12-08 by jeremiah-c-leary