jeremiah-c-leary/vhdl-style-guide #410: New Rule - Component vs Direct Instantiation

nfrancque, 2020-06-04:
Most codebases either decide to use component or direct instantiation exclusively.  Code reviews often have findings to make sure all are the same unless there are configurations involved.  

A simple implementation I think would be for VSG to flag component instantiations and/or declarations when configured to prefer direct, and vise versa for component.  

A more advanced implementation could automatically create a file some_ent_comp_pkg.vhd, declare the component, and change the instantiation to a component...or vise versa if direct is preferred.

The latter requires a high degree of knowledge about the codebase so could be overreaching.  The former is hopefully fairly straightforward.

jeremiah-c-leary, 2020-06-07:
funny...I thought there was already a rule for this, but there are only rules that check case.

jeremiah-c-leary, 2020-12-08:
So...what should the default be?  component or direct?

jeremiah-c-leary, 2020-12-08:
Rule is implemented.  Should it's severity be an error, warning or a guideline?

closed 2021-03-03 by jeremiah-c-leary