jeremiah-c-leary/vhdl-style-guide #411: Provide pre-configured set of commonly used configurations

nfrancque, 2020-06-04:
I would be nice to provide some alternative rulesets that may not be the tool's default suggestion, but are common enough to merit discussion. 

For instance, many people will simply not use the tool because a couple rules are against their own opinions.  Catering to these people fairly easily would greatly increase tool usage.

Some way of discovering these existing default configurations would be needed, as well as the ability to use them as a template for new user-defined ones.

jeremiah-c-leary, 2020-06-07:
I think issue #350 and pull request #397 cover how this could be done, but does not address what other rulesets should be there.  Currently a legacy (pre 2.0.0) ruleset and an indent ruleset are proposed.  I believe the biggest difference between legacy and 2.0.0 are some alignment and case rules.

Do you have any suggestions for other styles?

nfrancque, 2020-06-07:
I could name off a few things that are commonly disagreed on, but you're right it does seem difficult to come up with categories for each of them.  Would be nice if we could name them to provide some context.  Like catering to the personality most likely to use one set of conventions over another.

Some guesses -
- Indents, like you mentioned
- Capitalization of constants
- rising_edge vs. clk'event
- components vs. direct
- Maximum number of lines, signals, lines in a process...etc.  These are highly dependent on how you write code.  If your team prefers everything in one process your processes will be longer, that is not necessarily wrong.  Aware these aren't rules yet but could/may be soon?
- Signal or component naming conventions, some people always want U_ others do not

Any other ideas?

I think the trick is more in coming up with a set of rules that approximate discrete sets of users than thinking of rules people will disagree with - someone will always be around to disagree with something :smile:

jeremiah-c-leary, 2020-06-07:
There seems to be a vast number of possible styles.  We almost need a champion of a style.  For example, I imagine you have a preferred style.  Why not make that one available?

I just do not know how to gather more styles unless someone submits one.

It would be interesting if companies would submit their styles.  If that were to happen, VSG would almost become a universal code format translator.

ericjmcd, 2020-06-11:
Having a few pre-configured sets seems like a great idea to me.  But I agree, coming up with them would be difficult (and many).

One thing that might be workable would be a flag that creates a configuration based on the assumption that there are NO violations in a set of files.  i.e. Run VSG with the --output_configuration flag and another flag (--input_as_reference?) and enable all rules by default and any findings have those rules disabled in the generated output configuration file.

It might not be perfect, but should get someone about 90% of the way there.  I imagine most people have "gold" files that are examples of the desired style.  Using these as input, you could get something close.

nfrancque, 2020-06-11:
That is a fantastic idea!  If your golden reference i.e. still needs some lines aligned or small stuff, can iteratively tweak it until that file matches how you want it to look.  

It'd be important that VSG search for rules whose inverse match the current style, rather than disabling if the default doesn't match.  A rule should only be disabled if it finds no candidate rule configuration for the existing style - and that itself is probably a spark for discussion on a rule configuration to match the style.

jeremiah-c-leary, 2020-06-11:
> One thing that might be workable would be a flag that creates a configuration based on the assumption that there are NO violations in a set of files.

That is a borderline genius idea.

A lot of details to go over as @nfrancque pointed out.

nfrancque, 2020-07-19:
So it looks like this has devolved into two separate issues.

1) Providing common configurations and an ability for users to add more preset configurations to the tool.

2) Adding a feature to export a configuration that best matches the code being analyzed - possibly an entire codebase rather than a single file?

For 1, this was implemented by #397 and seems to be working now.  I would like to consider this done and close this ticket pending the result of the following request:

The names are not likely to be descriptive enough to capture what the intent of that style is.  I would propose adding a `description` field to the style yaml at the same level as the `name` field, and then display the description in the help dialog along with the name.  Users could then input something like "This style is identical to some_other_style, except that initial assignments to signals and variables is allowed.  This is intended to be used for testbench VHDL."

When creating my own style I found myself gravitating towards nonsensical names which are often catchier than descriptive ones - say, happy_armadillo - so that I could remember which one I picked by name, and verify with the description.

Thoughts?

For 2, it may be worth opening a separate issue after this is closed to discuss how that feature would work.