jeremiah-c-leary/vhdl-style-guide #412: xml produced by opition -j/--junit need to cast special characters

staerz, 2020-06-04:
**Environment**
fedora 30, VSG 1.10.0

**Describe the bug**
The produced xml file can happen to not be xml compliant due to characters like ">" or "<" in the failure description field.

This prevents tools like gitlab to parse this file when attempting to produce the pipeline report.

**To Reproduce**
Steps to reproduce the behavior:
1. Write code that violates concurrent_002, concurrent_004, sequential_002 or sequential_003
2. Run VSG on this code with the JUNIT option set
3. Have the produced XML file checked by an XML validator
4. See error

**Expected behavior**
It appears that the error messages as dumped to std_err are simply dumped into the "failure" without further treatment.

Special characters like "<" or ">" must be properly casted into "&lt;" and "&gt;".

See full list here: https://en.wikipedia.org/wiki/List_of_XML_and_HTML_character_entity_references#Predefined_entities_in_XML

**Additional context**
Here is an example section of invalid xml produced by VSG 1.10.0:
```xml
<?xml version="1.0" ?>
<testsuite errors="0" hostname="myhost" failures="140" timestamp="2020-06-04T02:47:48" tests="41" time="0" name="vhdl-style-guide">
  <properties>
  </properties>
  <testcase name="mysource" time="0" classname="failure">
    <failure type="Failure">
      concurrent_002: 111 : Remove all but one space after the <=.
      concurrent_002: 112 : Remove all but one space after the <=.
      concurrent_002: 113 : Remove all but one space after the <=.
      concurrent_004: 111 : Add a single space before the <=.
      concurrent_004: 112 : Add a single space before the <=.
      concurrent_004: 113 : Add a single space before the <=.
      sequential_002: 107 : Ensure a single space exists after the "<=" keyword.
      sequential_003: 52 : Ensure a single space exists before the "<=" keyword.
    </failure>
  </testcase>
  <system-out>
  </system-out>
  <system-err>
  </system-err>
</testsuite>
```

Viewing this file in Sublime, the `<=` is highlighted as a syntax error.

Checking with an XML validator reports the same issue:
https://www.w3schools.com/xml/xml_validator.asp

staerz, 2020-06-04:
Hi @jeremiah-c-leary,

as I reported to you, the option of the JUnit output is one reason why I chose VSG for our project.

Now with some "crappy" vhdl files not respecting any rules, I ran into that issue (it took me a minute to come to the bottom of it).

So if you could make fixing this (in my eyes) serious (but easy to fix) bug a priority, that would help us a lot!

staerz, 2020-06-04:
Hi @jeremiah-c-leary,

and just another question: Can I donate for the work you do here? If so, how?

jeremiah-c-leary, 2020-06-06:
Hey @staerz,

>So if you could make fixing this (in my eyes) serious (but easy to fix) bug a priority, that would help us a lot!

I will start working on this and get it into the 1.11.0 release

>and just another question: Can I donate for the work you do here? If so, how?

I appreciate the thought.  I would have to explore the options for donating.

jeremiah-c-leary, 2020-06-06:
Hey @staerz 

I just released version 1.11.0.  Could you please validate it resolves this issue.

Thanks,

--Jeremy

staerz, 2020-06-06:
Hi Jeremy,

after an update (sudo pip install vsg --upgrade) I run into this error:
```
[me@mymachine]$ vsg -v
Traceback (most recent call last):
  File "/usr/bin/vsg", line 11, in <module>
    load_entry_point('vsg==1.11.0', 'console_scripts', 'vsg')()
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib/python2.7/site-packages/vsg/__main__.py", line 11, in <module>
    from . import rule_list
  File "/usr/lib/python2.7/site-packages/vsg/rule_list.py", line 7, in <module>
    from . import utils
  File "/usr/lib/python2.7/site-packages/vsg/utils.py", line 10, in <module>
    import line
  File "/usr/lib/python2.7/site-packages/vsg/line.py", line 2, in <module>
    from vsg import utils
ImportError: cannot import name utils
```

Euhm, looks like a clash with python 2.7!? Any idea what I should do?

jeremiah-c-leary, 2020-06-06:
I can work on getting python version 2.7 installed and see if I get the same error.

staerz, 2020-06-06:
Yeah, that would be great! I mean I have both versions installed: Python 2.7.17, Python 3.7.7, but I'm actually not a Python expert at all and wouldn't actually know how I could set VSG to link with Python 3. On the other hand, if you intend to support Python 2.7, it should work ... (but well, Python 2.7 is over life time ... so the decision is yours, of course.

Just let me know what I should do :)

jeremiah-c-leary, 2020-06-06:
my version of python 2.7.17 has slightly different errors than yours:

```monovhdl-style-guide$ python bin/vsg --version
Traceback (most recent call last):
  File "bin/vsg", line 12, in <module>
    from vsg.__main__ import main
  File "/home/jeremiah/projects/vhdl-style-guide/bin/../vsg/__main__.py", line 11, in <module>
    from . import rule_list
  File "/home/jeremiah/projects/vhdl-style-guide/bin/../vsg/rule_list.py", line 7, in <module>
    from . import utils
  File "/home/jeremiah/projects/vhdl-style-guide/bin/../vsg/utils.py", line 10, in <module>
    import line
  File "/home/jeremiah/projects/vhdl-style-guide/bin/../vsg/line.py", line 2, in <module>
    from vsg import utils
ImportError: cannot import name utils
```
> On the other hand, if you intend to support Python 2.7, it should work ... (but well, Python 2.7 is over life time ... so the decision is yours, of course.

My work is still using 2.7, so I need to figure this out.  Importing is always confusing for me.

jeremiah-c-leary, 2020-06-06:
so This is interesting...I wonder if it is a circular reference:

utils.py import line while line import utils

jeremiah-c-leary, 2020-06-06:
k, I think I got it...Let me get a new version out for you.

jeremiah-c-leary, 2020-06-06:
@staerz  I just pushed version 1.11.1, could you check if the import error is fixed?

staerz, 2020-06-06:
Yup, nice, that works! And also the bug seems gone, I see the xml special chars!

And I confirm that the produced output is now properly parsed by gitlab!

Thank you very much for addressing this issue so quickly! I leave the pleasure of closing this issue to you.

jeremiah-c-leary, 2020-06-07:
Awesome.  I will close the ticket.

closed 2020-06-07 by jeremiah-c-leary