jeremiah-c-leary/vhdl-style-guide #413: Configuration of case for architecture_009 is ignored

Malcolmnixon, 2020-06-05:
**Environment**
This is occurring in VSG 1.10.0 on Windows

**Describe the bug**
No matter how I try setting the case to 'upper' (either global or for architecture_009: VSG reports:
architecture_009 | 9 | Ensure "end" and "architecture" keywords are lower case.

**To Reproduce**
[architecture_009.zip](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/4738200/architecture_009.zip)

1. Expand the attached zip containing the configuration of:
    rule:
      global:
        case : 'upper'
      architecture_009:
        case : 'upper'
2. Execute: vsg.exe -c architecture_009.yaml -f architecture_009.vhd
3. Observe the warning

**Expected behavior**
I would expect there to be no warning, as the config requests uppercase for everything including the END ARCHITECTURE line; and the VHD file contains:
END ARCHITECTURE SIM;

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

jeremiah-c-leary, 2020-06-07:
Greetings @Malcolmnixon ,

Unfortunately, the 1.x releases do not support configuring the case.  We upgraded some of the underlying reporting structure in the code and made it incompatible with the 1.x branch.

We are currently in the process of getting the 2.0.0 release done, which is taking a little longer than we have expected.  I like to think it is imminent, but it has been imminent for about two months.

If you would like the ability to change the case, and get a lot of new features, I would suggest following the directions on downloading the most up to date code and installing it:

[image]

We will be switching to this eventually, hopefully soon.

BTW, there is a method in 1.x to check a rules attributes that can be configured.  If you use the `-oc` option and give a file name, VSG will output every rule and it's configuration:

```json
     "architecture_009": {
       "disable": false,
       "fixable": true,
       "indentSize": 2,
       "phase": 6
     },
```

The code on master, and the future 2.0.0, adds the `-rc` command which allows you to request the configuration of single rule:

```mono
vhdl-style-guide$ python bin/vsg -rc architecture_009
{
  "rule": {
    "architecture_009": {
      "phase": 6, 
      "case": "lower", 
      "fixable": true, 
      "disable": false, 
      "indentSize": 2
    }
  }
}
```
As you can see in the second configuration, `case` is an option that can be set.

m-kru, 2020-06-23:
This issues should be closed when https://github.com/jeremiah-c-leary/vhdl-style-guide/pull/430 is merged.

jeremiah-c-leary, 2020-07-18:
Hey @Malcolmnixon ,

I just release version 2.0.0.  Could you check if this issue is resolved to your satisfaction?  If so, could you close this ticket.

Thanks,

--Jeremy

Malcolmnixon, 2020-07-18:
Just tested this using version 2.0.0 and it works fine. Many thanks and good job with the project!

closed 2020-07-18 by Malcolmnixon