jeremiah-c-leary/vhdl-style-guide #417: Add JSON output format for violations

jeremiah-c-leary, 2020-06-07:
To support integration with editors, outputting the violations in JSON format makes transferring data between applications easier.

This is in support of issue #401

qarlosalberto, 2020-06-08:
I'm testing the new features and changing my code. I get this json output:

```
{
  "colibri/test/sample.vhd": {
    "violations": [
      {
        "rule": "architecture_024",
        "linenumber": 28,
        "solution": "Add architecture name keyword."
      },
      {
        "rule": "entity_019",
        "linenumber": 15,
        "solution": "Add the entity name."
      },
      {
        "rule": "process_012",
        "linenumber": 22,
        "solution": "Add \"is\" keyword after the closing parenthesis."
      },
      {
        "rule": "process_016",
        "linenumber": 22,
        "solution": "Add a label for the process."
      },
      {
        "rule": "process_018",
        "linenumber": 25,
        "solution": "Add a label for the \"end process\"."
      }
    ]
  }
}
```

What is the the purpose of ** "colibri/test/sample.vhd"** ? What you do you think about?:

```
{
  "files" : 
  [
    {
      "file_path" : "/path/to/file",
      "violations": [
        {
          "rule": "architecture_024",
          "linenumber": 28,
          "solution": "Add architecture name keyword."
        },
        {
          "rule": "entity_019",
          "linenumber": 15,
          "solution": "Add the entity name."
        }
      ]
    },
    {
      "file_path" : "/path/to/file2",
      "violations": [
        {
          "rule": "architecture_024",
          "linenumber": 28,
          "solution": "Add architecture name keyword."
        },
        {
          "rule": "entity_019",
          "linenumber": 15,
          "solution": "Add the entity name."
        }
      ]
    }
  ]
}
```

jeremiah-c-leary, 2020-06-08:
I can update to that format.

jeremiah-c-leary, 2020-06-08:
I just pushed an update to match the above JSON format.  Could you update and double check?

Thanks,

--Jeremy

qarlosalberto, 2020-06-09:
Yes, its fine!

jeremiah-c-leary, 2020-06-21:
Awesome, I will close this issue.

closed 2020-06-21 by jeremiah-c-leary