jeremiah-c-leary/vhdl-style-guide #418: Enhance code tags with ability to disable a rule on the next line

jeremiah-c-leary, 2020-06-07:
Add the following to code_tags:

```vhdl
-- vsg-disable-next-line <rule>
```

This would disable a rule for the next line only.  Multiple rules could be listed also.

This issue in in support of issue #401

qarlosalberto, 2020-06-08:
I have checked how the others linters do it. They disable N rules in N comment lines. For example:

```
      // eslint-disable-next-line semi
      // eslint-disable-next-line radix
      let linter = new Linter.Linter(linter_name)
```

For the editor is easier in separate lines. Is it a problem for you? If it's a problem I can look for other solution.

jeremiah-c-leary, 2020-06-08:
I think I can support that. I could just collect exceptions until I hit a line that is not an exception.

jeremiah-c-leary, 2020-06-08:
Hey @qarlosalberto ,

I just pushed an update to the issue-401 branch.  Could you update and let me know if this update works for you.

Thanks,

--Jeremy

qarlosalberto, 2020-06-11:
Yes! I have added in the extension the ability to use a configuration file.
The configuration rules are a powerful feature.

jeremiah-c-leary, 2020-07-29:
Hey @qarlosalberto ,

I just committed a fix for next line code tags being flagged for non alignment.  Could you update and let me know if it works for you?

Thanks,

--jeremy

qarlosalberto, 2020-08-05:
@jeremiah-c-leary sorry for the delay, due to the holidays I'm not programming much in my free time.
Let me a time to test it :)

qarlosalberto, 2020-08-18:
It works fine between **architecture** and the **begin**:

[image]

But not after:

[image]

qarlosalberto, 2020-10-18:
Hi @jeremiah-c-leary do you plan to add the ```--all_phases --js`` argumentos to a release?

jeremiah-c-leary, 2020-10-27:
Hey @qarlosalberto ,

The answer is yes.  I am working on re-writing the rules for my new parser.  Only 30 more to go!!!  Unfortunately the new method broke the code tags, so I need to rethink how to implement it.  It shouldn't be hard though.

So once I finish re-writing the rules and hardening the new parser, hopefully by mid November I will incorporate the changes we discussed into the master branch.

Regards,

--Jeremy

jeremiah-c-leary, 2020-12-07:
@qarlosalberto ,
I just migrated this feature into master.

Could you pull down the latest master and let me know if it is working for you.

Thanks,

--Jeremy

jeremiah-c-leary, 2021-03-16:
Hey @qarlosalberto ,

I am going to close this issue for release 3.0.0 similar to the other issue I just closed. If there are any updates regarding this we can open a new issue.

--Jeremy

closed 2021-03-16 by jeremiah-c-leary