jeremiah-c-leary/vhdl-style-guide #419: Allow VSG to fix individual errors

jeremiah-c-leary, 2020-06-07:
There is a use case for editor integration when a single violation would need to be fixed.  Currently VSG will fix all violations it finds.

This feature request is to add the ability for VSG to fix individual errors.

This issue is in support of issue #401

qarlosalberto, 2020-06-08:
It would be nice to have a way to fix all errors of a type. It will be implemented in Visual Studio Code as "Fix all [code] problems". An possible idea is to pass a negative line:

code-of-problem-1 : -1
code-of-problem-2: 21

The first will fix all problems of type 1 and the second only the problem 2 in the line 21.

jeremiah-c-leary, 2020-06-08:
if we pass a JSON file in, we could use the following format:
```json
{
  "fix": {
     "rule": {
         "process_003": 3
     }
  }
}
```
and if you want to fix all of that rule:
```json
{
  "fix": {
     "rule": {
         "process_003": "all"
     }
  }
}
```
Would something like that work?

qarlosalberto, 2020-06-09:
perfect!

jeremiah-c-leary, 2020-12-07:
Hey @qarlosalberto ,

I just committed this feature to master.

Could you give a try to see what you think?

There is a caveat to using this feature which I documented.

Thanks,

--Jeremy

jeremiah-c-leary, 2021-03-16:
Hey @qarlosalberto,

I am going to close this issue for the 3.0.0 Release.  If you have any problems or other enhancements we can open another issue.

--Jeremy

closed 2021-03-16 by jeremiah-c-leary