jeremiah-c-leary/vhdl-style-guide #420: instantiation_011 crash: re.error: unbalanced parenthesis at position 3

joshrsmith, 2020-06-08:
**Environment**
```
Python 3.7.2, Linux 64-bit
vsg 1.11.1
```

**Describe the bug**
VSG Crashes with the following stack trace:

```
INFO: Analyzing rule: instantiation_011
INFO: Fixing rule: instantiation_011
Traceback (most recent call last):
  File "<PATH>/bin/vsg", line 10, in <module>
    sys.exit(main())
  File "<PATH>/lib/python3.7/site-packages/vsg/__main__.py", line 203, in main
    oRules.fix(commandLineArguments.fix_phase)
  File "<PATH>/lib/python3.7/site-packages/vsg/rule_list.py", line 137, in fix
    oRule.fix(self.oVhdlFile)
  File "<PATH>/lib/python3.7/site-packages/vsg/rule.py", line 73, in fix
    self._fix_violations(oFile)
  File "<PATH>/lib/python3.7/site-packages/vsg/rules/case_rule.py", line 75, in _fix_violations
    fix_function(oFile.lines[iLineNumber], word)
  File "<PATH>/lib/python3.7/site-packages/vsg/fix.py", line 88, in upper_case
    utils.change_word(oLine, sKeyword, sKeyword.upper())
  File "<PATH>/lib/python3.7/site-packages/vsg/utils.py", line 107, in change_word
    tLine = re.subn(r'\b' + sWord + r'\b', sNewWord, sLine, iMax, flags=re.IGNORECASE)
  File "<PATH>/lib/python3.7/re.py", line 203, in subn
    return _compile(pattern, flags).subn(repl, string, count)
  File "<PATH>/lib/python3.7/re.py", line 286, in _compile
    p = sre_compile.compile(pattern, flags)
  File "<PATH>/lib/python3.7/sre_compile.py", line 764, in compile
    p = sre_parse.parse(p, flags)
  File "<PATH>/lib/python3.7/sre_parse.py", line 944, in parse
    raise source.error("unbalanced parenthesis")
re.error: unbalanced parenthesis at position 3
```

**To Reproduce**
I have narrowed the crash down to a line similar to this:

`I_PORTMAP_ACTUAL         => I_ENTITY_INPUT(generate_loop_index)(function_call(G_GENERIC_A(generate_loop_index),8)-1 downto 0),`

This line is part of a port map statement, which is for an instance being generated in a generate loop.

If I comment the line out, then vsg continues on without any additional issues.

Note, I have another line of code that looks like this which does not trigger the problem:
`I_PORTMAP_ACTUAL_OK      => I_ENTITY_INPUT(generate_loop_index)(function_call2(G_GENERIC_A(generate_loop_index))-1 downto 0),`

I suspect the problem is the `function_call` which requires 2 arguments, whereas `function_call2` only has one argument.

**Expected behavior**
Crash would not occur and I don't believe there is anything to fix here.

**Screenshots**
N/A

**Additional context**
N/A

jeremiah-c-leary, 2020-06-08:
@joshrsmith ,

I suspect the first comma is considered the end of the port instantiation.

jeremiah-c-leary, 2020-06-09:
I almost have a fix for this.  I have a test that is failing due to my changes that I want to track down first.  This is issue #421 .

jeremiah-c-leary, 2020-06-20:
Hey @joshrsmith ,
This fix is scheduled for the 1.12.0 release.

jeremiah-c-leary, 2020-06-21:
Hey @joshrsmith,

I just release version 1.12.0.  Could you check that it fixes this issue.

Thanks,

--Jeremy

joshrsmith, 2020-06-23:
Confirmed resolved!

closed 2020-06-23 by joshrsmith