jeremiah-c-leary/vhdl-style-guide #423: Emacs support

csantosb, 2020-06-13:
**Is your feature request related to a problem? Please describe.**

I found no reference on how to include support for use under emacs.

**Describe the solution you'd like**

Calling "C-c M-b" in code below applies vsg to current region or paragraph. A prefix argument "C-u C-c M-b" applies to the whole buffer. Just needs to set the path to the config.yaml file.

**Describe alternatives you've considered**

Not found.

**Implementation**

Improvements welcomed ;-)

    (defun vhdl-vsg-beautify-region-or-buffer (arg)
      "Call vsg on current region or paragraph if region is not active.
    Whit a prefix argument, apply on whole buffer."
      (interactive "P")
      (if (equal arg '(4))
          ;; i. apply on current buffer
          (progn
            ;; apply vhdl style guide on file on disk
            (let ((vsg-config-file
                   (format "%s/vsg/config.yaml" "/HOME/USER/")))
              (when
                  (and (executable-find "vsg")
                       (file-exists-p vsg-config-file))
                (progn
                  (shell-command
                   (format "vsg -f %s -c %s --fix" (buffer-file-name) vsg-config-file))
                  (revert-buffer nil t)))))
        ;; ii. apply vhdl style guide on active region
        (let ((vsg-config-file
               (format "%s/vsg/config.yaml" "/HOME/USER"))
              (tmp_filename (make-temp-file "")))
          (when
              (and (executable-find "vsg")
                   (file-exists-p vsg-config-file))
            (save-excursion
              (when (not (region-active-p))
                (mark-paragraph))
              (write-region (region-beginning) (region-end) tmp_filename)
              (shell-command
               (format "vsg -f %s -c %s --fix" tmp_filename vsg-config-file))
              (delete-region (region-beginning) (region-end))
              (insert-file-contents tmp_filename)
              (delete-file tmp_filename))))))

    (define-key vhdl-mode-map (kbd "C-c M-b") #'vhdl-vsg-beautify-region-or-buffer)

m-kru, 2020-06-18:
This is rather not related to `vhdl-style-guide`. `vhdl-style-guide` tries to follow Unix philosophy. I do not use emacs, but I am sure there is a way to bind some program to do some stuff with the buffer.

csantosb, 2020-06-18:
Sure, just posted it here in case someone else wants to use the style guide from within emacs.

closed 2020-06-18 by csantosb