jeremiah-c-leary/vhdl-style-guide #424: Prefix rule for process and generate labels

marph91, 2020-06-14:
**Is your feature request related to a problem? Please describe.**
I would like to define prefixes for the process and generate labels.

**Describe the solution you'd like**
A rule for process, respectively generate, to define the prefixes.

m-kru, 2020-06-18:
Definitely feasible. It would be quite easy to implement, maybe you could try to make PR :)?

jeremiah-c-leary, 2020-06-20:
Interesting idea.  You could use an existing prefix rule as a template.  It would just be a matter of pulling the label and checking if it starts with the prefix.

marph91, 2020-06-21:
Sure, I will take a look. For my designs, the prefixes are mostly `PROC_` respectively `GEN_`. Should I use them as default or do you prefer something else?

m-kru, 2020-06-21:
They are fine as the defaults, but remember that by default everything should be lowercase, so it would be `proc_` and `gen_`.

jeremiah-c-leary, 2020-06-21:
Should this rule be on by default?

Also, I just committed some tokenizing code when I fixed a bug for issue #427.  It might make your checks easier.

m-kru, 2020-06-21:
This is something, that we already tried to discuss in some other thread. In my opinion all rules should be disabled by default, and enabling would be done in builtin styles and configuration files.

marph91, 2020-06-21:
@m-kru: Are you sure about lowercase? In the [docs](https://vhdl-style-guide.readthedocs.io/en/latest/process_rules.html#process-017) its written uppercase. Also when I tried it, the process labels were converted to uppercase.

@jeremiah-c-leary: I think it should be disabled, to keep it consistent to the other prefix and suffix rules.

m-kru, 2020-06-21:
@marph91 Yes, I am sure. This doc probably needs an update. We are leaning towards making everything lowercase by default, and then uppercase in style and configuration files. It is more consistent approach.

marph91, 2020-06-21:
That makes sense. Now I see that the case is checked by a separate rule.

Btw.: The [shebang of the vsg executable](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/3e987c7a3981473156a665dfb710ce337df86bd6/bin/vsg#L1), invoked on my system (Opensuse tumbleweed) Python 2, which led to strange errors. I had to fix it by writing explicitly `python3`. Should I include it in the PR? Not sure if it would break on other distros.

m-kru, 2020-06-21:
Do not include it in PR, Python 2 is no longer supported.

jeremiah-c-leary, 2020-06-23:
@marph91 ,
I thought the shebang was replaced when pip does an install.

marph91, 2020-06-23:
I didn't install via pip. I used the following lines to get the develop version:
```bash
git clone https://github.com/jeremiah-c-leary/vhdl-style-guide.git
python3 setup.py install
```
Problem is that `python` is an alias of `python2`. At least in Opensuse. Probably this isn't the case for other distributions, as you don't have the problem.

closed 2020-06-27 by jeremiah-c-leary