jeremiah-c-leary/vhdl-style-guide #425: More descriptive rule names

marph91, 2020-06-14:
**What is your question?**
I just started to use vsg in a [small project](https://github.com/marph91/yaaes/blob/master/vsg_config.yaml). When searching for rules it was a bit cumbersome to find the exact rule. Also the configuration file looks similar to this:

```yaml
rule:
  signal_007: # default initialization
    disable: true
  subtype_004: # prefix
    disable: false
  type_015: # prefix
    disable: false
```

I added the comments to remember what the rule is about. Would it be feasible to add an identification word to the rule name. For example `subtype_004_prefix` or `subtype_prefix` instead of `subtype_004`?

m-kru, 2020-06-18:
It is feasible, but will never be implemented (in my humble opinion), @jeremiah-c-leary. Such change would have some significant impact on the implementation, plus the gain is almost 0.
What is more, the needed change would make Python code more cumbersome.

`vsg` has really good documentation. Read it, find the rule, edit config file. Once you create your configuration you are probably going to stick with it.

Btw. With next release there will be predefined formatting styles to choose from.

jeremiah-c-leary, 2020-06-20:
Greetings @marph91,
I can see the desire to have more descriptive names.  I run into the same problem when creating configurations.  `signal_007` is not very descriptive.  You only know it has something to do with signals.

The intent was to drive people to the documentation, as it would hopefully be more descriptive than the rule title.

I would not want to change the format of the existing names for the reasons @m-kru has mentioned.

  There might be an another approach.  What if there as an alternative name for each rule.  For example, `signal_007` could have another attribute with a value of `signal_default_assignment`.  When added into a configuration file:
```yaml
rule:
   signal_default_assignment:
     disable : true
```
It would be more descriptive.  The question becomes how does the user learn about these alternative names and how do they correlate them back to the real rule name.

I suppose we could add them to the documentation:
signal_007 (signal_default_assignment).
signal_008 (signal_prefix)
signal_011 (signal_type_case)
signal_012 (align_multiple_signals_on_a_single_line)
signal_013 (align_colons_of_signal_assignments_in_architecture_declarative_region)
signal_014 (consistent_capitalization_of_signal_names)

For the examples above, I am just looking at the one sentence descriptions for the rules.

There would be issues to resolve when merging configurations.  For example, if you had one configuration with `signal_007` defined and one with `signal_default_assignment` defined there would have to be a lookup to translate `signal_default_assignment` into `signal_007`.

That could be done with a method on the `rule_list` class to search for the rule with a matching alternative name and then return that rule's ID.

Would this enhance the documentation?  For example:  currently readthedocs looks like this in the table of contents:  
[image]

but if we added the alternative names, it could look like this:

[image]

Updating 350+ rules would be daunting.

Comments?

marph91, 2020-06-21:
I agree, updating all the rules probably isn't worth the effort. When the configuration is fixed once, it doesn't matter anymore.

Maybe one thing that could be done with manageable effort is to update only the docs, not the python code. I. e. write a short identifier to each rule, like at your last image. This would speed up at least the use case when looking for a specific rule, where you know already the syntax. For example when looking for multiple prefix rules (signal, constant, variable, ...).

Do you think it would be useful?

staerz, 2020-06-22:
I my opinion, don't complicate things!

Keep the rules as they are, they follow a very nice and easy to understand convention, but sure, the numbering may appear arbitrary.

But does that matter?

My use case (and I have disabled some 38 rules for various reasons for now):
- apply VSG to your code base as VSG comes
- check (RTFM) the offending rules: If you agree to the rule, fix the problem in the source code, if you don't, disable the rule

This is a one-time action. So you look up what "rule_123" is, and that's it.

jeremiah-c-leary, 2020-06-23:
This is the eternal struggle.  The more I think about this idea, the more I like it, but as @staerz says:
> In my opinion, don't complicate things!

I follow the same methodology as @staerz, and yet I can see if I had to go back to previous configurations and saw:
```yaml
  rule:
    length_001 : 
        disable : true
```
I would have no idea what it is.  That would prompt me to go look at the documentation again.  Which is probably why @marph91 is putting comments in the YAML file to describe the rule does.

Let me do a quick poll at work and see where people land.  If you know of anyone else who could have some good input on this idea I would like to hear it.

m-kru, 2020-06-23:
I also use comments, example:
```yml
rule:
    global:
        indentSize: 2
    # Disable uppercase entity name
    entity_008:
        disable: true
    entity_012:
        disable: true
    architecture_014:
        disable: true
    # Disable uppercase port name
    port_010:
        disable: true
    # Disable uppercase architecture name
    architecture_011:
        disable: true
    architecture_013:
        disable: true
    # Disable Use 'event for clocks
    process_029:
        disable: true
    # Disable Enclose boolean expression in ()'s
    if_002:
        disable: true
    # Allow default assignment in port declaration
    port_012:
        disable: true
    # Allow constants to be uppercase
    constant_004:
        disable: true
    # Allow lowercase process label
    process_017:
        disable: true
    process_019:
        disable: true
    # Allow lowercase generate label
    generate_005:
        disable: true
    generate_012:
        disable: true
    # Allow lowercase for loop label
    for_loop_003:
        disable: true
    # Generics must start with G_
    generic_020:
        disable: false
```

Some of the alternative names would have to be quite long to express what the rule is about.
Going through the documentation to get to know the rules is not a time consuming process.

staerz, 2020-06-23:
@m-kru, I do it exactly like this as well!

marph91, 2020-06-23:
Ok, using comments is fine for me, considering the high efforts for the little gain. Imo the ticket can be closed if there is no new argument for the alternative names.

closed 2020-06-23 by m-kru