jeremiah-c-leary/vhdl-style-guide #426: Prefer "-f" over "file_list" in config file to allow individual analysis and fixing

ericjmcd, 2020-06-17:
When providing both "-f" and "-c" options to specify both a file and a configuration file, VSG ignores the "-f" file(s) and uses "file_list" from the config file.

Ideally, this would use the "-f" option as a user would not normally provide both unless they want to specifically analyze or fix a specific file.  I have been using file_list in my configuration file but when working on new files, I'd like to only analyze/fix one file at a time and this is not possible when file_list is used in the configuration file.

jeremiah-c-leary, 2020-06-20:
Hey @ericjmcd,

>When providing both "-f" and "-c" options to specify both a file and a configuration file, VSG ignores the "-f" file(s) and uses "file_list" from the config file.

I can see your point on this.  Another option is to merge the `-f` and `-c` files.

Have you tried splitting your configuration up.  You can have one configuration be the rules: 
 `my_rules.json`
```json
{
  "rules": {
     "entity_001": {
         "disable": true
     }
  }
}
```
...and another for the files: `project_files.json`
```json
{
  "file_list": [
    "entity1.vhd",
    "entity2.vhd",
    "architecture1.vhd"
  ]
}
```
Then you can call both of them on the command line:
```mono
$vsg -c my_rules.json project_files.json
```
VSG will merge the two configurations, with the configuration to the right having priority.
This would allow you to run on a single file by using this command:
```mono
$vsg -f single_file.vhd -c my_rules.json
```
Then you could just add it to the `project_files.json` configuration when you are satisfied it goes through VSG.

jeremiah-c-leary, 2020-06-20:
I thought I had documented the merging of configurations.  It must have been deleted somewhere along the way.  I will try to find it and update the documentation.

ericjmcd, 2020-06-22:
Thanks - that works like a charm.  I might be useful to throw a warning if "file_list" and "-f" are both used.  I'll leave this ticket open for you to track updating the documentation.

jeremiah-c-leary, 2020-06-23:
Well...I found the documentation.  It was under the Usage section, which doesn't seem correct.  I think I will move it.

closed 2020-06-23 by jeremiah-c-leary