jeremiah-c-leary/vhdl-style-guide #427: Incorrectly fixes one-line multi-assignment port mappings.

ericjmcd, 2020-06-18:
**Environment**
Python3.6
VSG 1.11.1

**Describe the bug**
VSG incorrectly fixes one-line multi-assignment port mappings.

**To Reproduce**
This snippet of code
```
    inst: MODULE
    port map (
        A => a, B => b, C=>c);

    inst2: MODULE
    port map (
        A => a, B => b, C=>c
        );
```
becomes
```
    inst : module
        port map (
            a => a,

            b => b,

    inst2 : module
        port map (
            a => a,
            b => b
        );
```
**Expected behavior**
```
    inst : module
        port map (
            a => a,
            b => b,
            c => c
        );

    inst2 : module
        port map (
            a => a,
            b => b,
            c => c
        );
```

jeremiah-c-leary, 2020-06-20:
I thought this would be a quick fix, but it is taking a little longer.

jeremiah-c-leary, 2020-06-21:
Hey @ericjmcd,

I just release version 1.12.0.  Could you valid that version fixes this issue.

Thanks,

--Jeremy

ericjmcd, 2020-06-22:
Appears fixed in 1.12.0.  Thanks once again for the quick turn-around!

closed 2020-06-22 by ericjmcd