jeremiah-c-leary/vhdl-style-guide #431: instantiation_024 false positive with multi-line actual in generic map

joshrsmith, 2020-06-23:
**Environment**
VSG 1.12.0
Python 3.7.2

**Describe the bug**
I found another edge case, similar to #407 that this issue is triggered.

In a generic map:
```
G_FORMAL    => G_ACTUAL(C_A(i) + C_B(i) - 1
                               downto C_C(i)),                           -- triggered on this line
```
Note that the actual is broken up over two lines because the constants are very long names. Also, there are a few generic assignments in the portmap similar to this in a row. The second line of each assignment is what gets flagged by vsg (as indicated with `--triggered` comment above)

I get the VSG output: "Rewrite positional assignment to explicit assignment."

**To Reproduce**
See above

**Expected behavior**
This coding style should not cause this failure.

**Screenshots**
N/A

**Additional context**
N/A

jeremiah-c-leary, 2020-07-18:
Hey @joshrsmith ,

I just realized this was 24 days old.  I am planning to release version 2.0.0 this weekend and then I can focus on this bug.

jeremiah-c-leary, 2020-12-05:
Hey @joshrsmith ,

I apologize for taking so long to address this.  I rewrote my parser to be more robust so it would not get confused when I was using regular expressions.

I copied your code snippet and ran it against what I have in master.  The good news is the new parser can handle the above code without throwing the error you mentioned.  I am assuming it is `instantiation_024`.

Could you try your code using master?

Otherwise, I will plan on closing this issue when I release 3.0.0

Thanks,

--Jeremy

jeremiah-c-leary, 2021-01-11:
Hey @joshrsmith ,

Just wanted to circle back on this issue and see if it is still a problem.

--Jeremy

joshrsmith, 2021-01-11:
I believe it has been resolved in 3.0 development. Although the original rule, instantiation_024 has been deprecated, I am not getting false positives on generic_map_008 or port_map_008 now.

closed 2021-01-11 by joshrsmith