jeremiah-c-leary/vhdl-style-guide #440: Don't care in port map generates errors

Emilv2, 2020-07-16:
**Environment**
vsg 1.12.0

**Describe the bug**
Don't cares in port map are treated wrongly as comments or math operators

**To Reproduce**

input file:
    
    -- test.vhd
    
    library UNISIM;
      use UNISIM.vcomponents.all;
    
    entity TEST is
      port (
        INPUT   : in    std_logic;
        DI0     : in    std_logic;
        OUTPUT  : out   std_logic
      );
    end entity TEST;

    architecture RTL of TEST is

      INV_3_CARRY_0 : CARRY4
        port map (
          CO             => INPUT,
          O              => OUTPUT,
          CI             => '-',
          CYINIT         => '-',
          DI(3 downto 1) => "---",
          DI(0)          => s4,
          S              => "1110"
            );

    end architecture TEST;

output file:

    -- test.vhd
    
    library UNISIM;
      use UNISIM.vcomponents.all;
    
    entity TEST is
      port (
        INPUT   : in    std_logic;
        DI0     : in    std_logic;
        OUTPUT  : out   std_logic
      );
    end entity TEST;

    architecture RTL of TEST is

      INV_3_CARRY_0 : CARRY4
        port map (
          CO             => INPUT,
          O              => OUTPUT,
          CI             => '-',
          CYINIT         => '-',
          DI(3 downto 1) => "
          DI(0)          => s4,
          S              => "1110"
            );

    end architecture RTL;

vsg commandline output:

    File:  test.vhd
    ==============
    Phase 1... Reporting
    Phase 2... Reporting
      whitespace_011            |         20 | Add a single space before and/or after math operator.
      whitespace_011            |         21 | Add a single space before and/or after math operator.
    Phase 3... Not executed
    Phase 4... Not executed
    Phase 5... Not executed
    Phase 6... Not executed
    Phase 7... Not executed
    ==============
    Total Rules Checked: 151
    Total Violations:    2

**Expected behavior**

    CI             => '-',
    CYINIT         => '-',

should not generate `Add a single space before and/or after math operator.` message

     DI(3 downto 1) => "---",

should not turn into  `DI(3 downto 1) => "`

jeremiah-c-leary, 2020-07-16:
Hey @Emilv2 ,

I will schedule this for the 1.13.0 release.

jeremiah-c-leary, 2020-07-18:
Hey @Emilv2 ,

Same comment that I put into #441 .  I will schedule this for the 2.1.0 release.

jeremiah-c-leary, 2020-12-05:
Hey @Emilv2 ,

It has been a long time.

I have finished by re-write of my parser and wanted to circle back to this issue.  I copied your code snippet and ran it against what I have in master.  It did not throw the `whitespace_011` errors.  It also looks like the `DI(3 downto 1) => "` issue is also resolved.

Could you validate master resolves this issue for you.

If not, I will plan on closing this issue when I release 3.0.0.

Thanks,

--Jeremy

Emilv2, 2020-12-15:
Seems to be fixed in master, thanks!

closed 2020-12-15 by Emilv2