jeremiah-c-leary/vhdl-style-guide #441: comments are treated as code

Emilv2, 2020-07-16:
**Environment**
vsg  1.12.0 from pypi

**Describe the bug**
comments are treated as code and get "fixed" too.

**To Reproduce**
Steps to reproduce the behavior:

input file:

    -- test.vhd

    library UNISIM;
      use UNISIM.vcomponents.all;

    entity TEST is
      port (
        INPUT0   : in    std_logic;
        INPUT1   : in    std_logic;
        INPUT2   : in    std_logic;
        INPUT3   : in    std_logic;
        OUTPUT   : out   std_logic
      );
    end entity TEST

    architecture RTL of TEST is

      LUT : LUT4
        --generic map (INIT => "01010101010101010101010101010101")
          generic map (
            INIT => "0101010101010101"
          )
          port map (
            O  => OUTPUT,
            I0 => INPUT0,
            I1 => INPUT1,
            I2 => INPUT2,
            I3 => INPUT3
          );

    end architecture RTL;

output file:

    -- test.vhd

    library UNISIM;
      use UNISIM.vcomponents.all;

    entity TEST is
      port (
        INPUT0   : in    std_logic;
        INPUT1   : in    std_logic;
        INPUT2   : in    std_logic;
        INPUT3   : in    std_logic;
        OUTPUT   : out   std_logic
      );
    end entity TEST

    architecture RTL of TEST is

      LUT : LUT4
          --generic map ( (INIT => "01010101010101010101010101010101")
          generic map (
            INIT => "0101010101010101"
          )
          port map (
            O  => OUTPUT,
            I0 => INPUT0,
            I1 => INPUT1,
            I2 => INPUT2,
            I3 => INPUT3
          );

    end architecture RTL;

` vsg -f test.vhd --fix  ` output:

    File:  test.vhd
    ===============
    Phase 1... Reporting
    Phase 2... Reporting
      instantiation_025         |         19 | Move the ( to the same line as the "generic map" keyword.
    Phase 3... Not executed
    Phase 4... Not executed
    Phase 5... Not executed
    Phase 6... Not executed
    Phase 7... Not executed
    ===============
    Total Rules Checked: 151
    Total Violations:    1

**Expected behavior**

    --generic map (INIT => "01010101010101010101010101010101")

this is a comment and should not be changed to

    --generic map ( (INIT => "01010101010101010101010101010101")

jeremiah-c-leary, 2020-07-16:
@Emilv2 ,

I agree, the commented out generic map line should have been ignored.
I will schedule this for  the 1.13.0 release.

jeremiah-c-leary, 2020-07-18:
Hey @Emilv2 ,

I am planning to release 2.0 this weekend unless something big prevents me.  So I was going to schedule this for the 2.1.0 release.

jeremiah-c-leary, 2020-07-28:
Hey @Emilv2 ,

I have a fix for this issue.  It will be in the 2.10 release.

closed 2020-10-08 by jeremiah-c-leary

Emilv2, 2020-10-09:
@jeremiah-c-leary Seems indeed to be fixed in the latest release, thanks! Good luck with your parser rewrite.