jeremiah-c-leary/vhdl-style-guide #443: Check signal_013 also in block and process environment

staerz, 2020-07-17:
**Is your feature request related to a problem? Please describe.**
The existing rule `signal_013` checks for the proper alignment of the colon in signal declaration sections.
This check seems to be limited to the bare signal declaration section of the `architecture`.

VHDL allows to declare signals also in `block` and `process` environments (as it also allows the definition of (local) constants and variables there).

Currently, the rule `signal_013` does not complain on misaligned colons in these sections.

See the following (not working) code example:

```
architecture behavioral of counter is

-- here VSG would complain
  signal counter    : std_logic_vector(15 downto 0);
  signal counter_2  : std_logic_vector(15 downto 0);

begin

  new_block: block
-- here VSG wouldn't complain
    signal counter : std_logic_vector(15 downto 0);
    signal counter_2  : std_logic_vector(15 downto 0);
  begin
    counter_sctrl <= counter_sctrl_2;
  end block;

  new_process : process (clk)
-- here VSG wouldn't complain
    signal counter_sctrl : std_logic_vector(15 downto 0);
    signal counter_sctrl_2  : std_logic_vector(15 downto 0);
  begin
    counter_sctrl <= counter_sctrl_2;
  end process;
  ...
```

**Describe the solution you'd like**
The rules applied for signal declarations should also be applied in the `block` and `process` environment.
It is possible that more than just rule `signal_013` are affected, I did not check.

It also appears that there are no rules for the `block` environment at all.
I.e. I would get
"process_024               |         37 | Ensure a single space exists between process label and :"
but not the relative complaint for the block section.
(Maybe that's for an independent issue...)

**Describe alternatives you've considered**
There is no alternative :)

**Additional context**
Let's make VSG great again.

jeremiah-c-leary, 2020-07-18:
I agree, the rule should operate on any signal declaration.  I might make separate rules based on the location to allow for finer control of the formatting.

> It also appears that there are no rules for the block environment at all.
I.e. I would get
"process_024 | 37 | Ensure a single space exists between process label and :"
but not the relative complaint for the block section.
(Maybe that's for an independent issue...)

Blocks are currently a blind spot for VSG.  I think it will only check the indent.  After the 2.0 release I can add some rules for blocks.

> Let's make VSG great again.

Love it!!

staerz, 2020-07-20:
Yes, the rule should operate on any signal declaration, but I don't see why you'd need new rules for different levels.

I see that you removed `signal_013` in VSG 2.0.0?!

I mean `signal_013` was meant to "This rule checks the colons are aligned for all signals in the architecture declarative region." - if you simply remove the "architecture" from that description, it would still be fully correct!

jeremiah-c-leary, 2020-12-10:
Hey @staerz ,

I did some digging and the rule `process_033` should cover the alignment of colons in the process declarative region.

Also, the new block rule `block_401` should cover the alignment of colon in the block declarative region.

Could you double check the current master addresses this issue?

Thanks,

--Jeremy

staerz, 2020-12-17:
I confirm that `process_033` is complaining (and sorry, the example is actually wrong, one may not define signals there, but only variables).

For some reason though, it seems VSG doesn't check for block rules - do I have to enable them first?

jeremiah-c-leary, 2020-12-17:
> For some reason though, it seems VSG doesn't check for block rules - do I have to enable them first?

I forgot to enable them.  I had already pushed a fix as part of issue #457

staerz, 2020-12-17:
Alright, after a new pull I see the block rules get checked and I confirm that they cover the cases above (namely `process_033` and `block_401` as you write).

I'd say another one to be closed :wink:

jeremiah-c-leary, 2020-12-17:
Sweet....getting closer to the 3.0.0 release.

--Jeremy

closed 2020-12-17 by jeremiah-c-leary