jeremiah-c-leary/vhdl-style-guide #444: VHDL-2008 (if|case) generate and blocks

staerz, 2020-07-17:
**Environment**
VSG 1.11.1, Fedora 32

**Describe the bug**
:bug: indentation in/after `else generate` and `case generate` (concurrent_001, generate_001, generate_006, generate_007, generate_016, signal_001) (VSG 1.11.1)

VSG seems to have a problem with the VHDL-2008 syntax for `else generate` and `case generate` leading to a fake `Invalid indentation.` warnings.

The following code
```vhdl
  gen_ttcrx : if <boolean expression> generate
  begin
    input  <= <something>;
    output <= <something else>;
  else generate
    <block label> : block
      signal intermediate        : <some type>;
    begin
      intermediate <= <pre_something>;
      input <= <some function of intermediate>

      <instance label> : entity <library>.<entity>
      generic map (
        ...
      )
      port map (
        input  => input,
        output => output
      );

    end block;

  end generate;
```

or alternatively the following code

```vhdl
  coef_int_gen : case SEL_COEF generate
    when 1 =>
    -- select coefficient of 1
      c_i <= C_ONE;
    when 2 =>
      -- failure message when N_DEPTH less than 6 for Weiner filter

      msg_gen : if N_DEPTH < 6 generate
        assert not (N_DEPTH < 6 and SEL_COEF = 2)
          report "The Wiener filter MUST have N_DEPTH > 6. The current N_DEPTH : " & integer'image(N_DEPTH) & "!" & CR
          severity failure;
      -- select coefficient of Weinier filter
      else generate
        c_i <= (0 to 5 => C_WIENER, others => 0);
      end generate;

    when others =>
      --null;
  end generate;
```

raises these fake warnings:
```
  generate_016              |        190 | Invalid indentation.
  generate_006              |        352 | Invalid indentation.
  generate_007              |        385 | Invalid indentation.
  generate_001              |        389 | Invalid indentation.
  signal_001                |        390 | Invalid indentation.
  concurrent_001            |        426 | Invalid indentation.
```
If the internal `if...generate...` in the latter example is removed, then VSG can pass.

Running VSG with the `--fix` option results in wrongly indented code.

**To Reproduce**
Run VSG on the code examples above.

**Expected behavior**
The rules concurrent_001, generate_001, generate_006, generate_007, generate_016, signal_001 should not flag as violated in the circumstances as given in the code examples above.

**Screenshots**
N/A

**Additional context**
N/A

jeremiah-c-leary, 2020-07-18:
Hey @staerz ,

I am planning to release 2.0 this weekend unless something untoward happens.  I will the schedule this for the 2.1 release.

staerz, 2020-07-20:
Hi @jeremiah-c-leary,

yes, I'm just updating to VSG 2.0.0. It changes a few things and I get many complaints on `entity_017` now - I'll have to see what that is :)

Incorporating VHDL-2008 syntax in VSG for us is utterly important as we use it quite often (simply the `if ... generate ... else generate` and `case ... generate` is too awesome to not use it!).

jeremiah-c-leary, 2020-09-23:
Hey @staerz ,

I copied your code to a file and ran my interim VSG on it with the new parser:

[image]

I still need to refactor the generate rules though.

staerz, 2020-09-23:
Hi @jeremiah-c-leary,

very good to see that a solution is in sight :+1:

jeremiah-c-leary, 2020-12-05:
@staerz ,

I believe the new parser can handle this properly now.  Could you validate it by pulling down master?

Thanks,

--Jeremy

staerz, 2020-12-10:
I re-enabled concurrent_001 and am sorry to tell you that still some indentation rules are doing some fake reporting.

I do though see an improvement and these fake reports don't seem to be related to the VHDL-2008 `if|case generate`, but to `block` statements. You seem not to like `blocks` :wink:

I have the following scenario where I still get the fake error:

```vhdl
architecture behavioral of fpga_registers is

begin

  blk_fpga_avmm_registers : block

    --! Uptime as 64 bit unsigned
    signal fpga_uptime   : std_logic_vector(G_FPGA_UPTIME_W-1 downto 0);
    ...
  begin
    -- let's simulate some "no downtime"
    fpga_downtime <= (others => '0');
    ...
  end block;

end architecture behavioral;
```

Any statement inside the block which is not an entity instantiation (maybe that's because I only uncommented "concurrent_001" now) is being reported.
If I move everything inside the block one level to the left, the errors are gone.

Note that it doesn't change upon adding the optional keyword `is` after the `block`.

jeremiah-c-leary, 2020-12-10:
It doesn't look like I have any indenting for blocks yet.  Let me work on issue #457 and that should clean up these concurrent errors.

jeremiah-c-leary, 2020-12-10:
Hey @staerz ,

I just committed the block rules.  Could you validate the latest master fixes this issue.

Thanks,

--Jeremy

staerz, 2020-12-17:
Hi @jeremiah-c-leary,

I reenabled these offending rules and could not find a problem (at least not on a solid subtree of our project).

Feel free to close that issue then.

jeremiah-c-leary, 2020-12-17:
Woot!!  Closing the issue.

closed 2020-12-17 by jeremiah-c-leary