jeremiah-c-leary/vhdl-style-guide #445: Entity map indentation

staerz, 2020-07-17:
**Is your feature request related to a problem? Please describe.**
It's a matter of convention, but it would be very nice to have maybe a configuration to chose the one or the other, see description.

**Describe the solution you'd like**
VSG currently would indent as follows:

```vhdl
some_block : block
begin
  some_entity : entity misc.me
    -- here we go for the next layer of indentation
    generic map (
      ...
      last_port => last_port
    );
    -- here is where the one layer too much of indentation is in my eyes.
    -- because a signal assignment would be here:
  some_signal <= this_signal;
end block;
```

while it should rather be
```vhdl
some_block : block
begin
  some_entity : entity misc.me
  generic map (
    ...
    last_port => last_port
  );
end block;
```

That example is written in a `block` environment, but applies to all environments.
It basically reduces down to go for

```vhdl
  some_entity : entity misc.me
    generic map (
      ...
    );
  next_statement <= ...
```
or (which I would prefer)
```vhdl
  some_entity : entity misc.me
  generic map (
    ...
  );
  next_statement <= ...
```

**Describe alternatives you've considered**
The alternative is to live with the (double) indentation that VSG suggests, but that really looks odd in my eyes.

**Additional context**
N/A

jeremiah-c-leary, 2020-07-18:
Hmmm....This is a different kind of configuration since indenting is fixed by the code that analyzes the VHDL before the rules are applied.

Maybe something like this:

```yaml

  indenting:
   library : 0
   use : 1
   signal : +1
   entity : 0
   generic : -1
```
...where just a number sets an absolute indent value.  The `+1` would indicate to indent an extra level.  The `-1` would indicate to un-indent one level.

So in your case you would use:

```yaml
  indenting:
    generic_map : -1
```

When the analyzer goes through the code it would normally increment the indent level by 1 and then assign it to that line.  If it was set to `-1`, then it would not indent that line.  everything else would then indent based on that lines indent level.  Essentially everything would shift to the left.  I will have to watch out for the end of the generic map and ensure I do not un-indent again.  

A side benefit of this is that it gives me the opportunity to improve the indenting algorithm.

staerz, 2020-07-20:
Yes, that sounds like a very good solution!

staerz, 2020-12-19:
Hi @jeremiah-c-leary,

did you have the chance to make any move on this issue?

This is actually cross-related with #502 as comments that I do for any port or generic will be thrown as errors on `comment_010` - that is very annoying :wink:

So until this issue here is still up, I'll still have to disable `comment_010`. :sob:

jeremiah-c-leary, 2020-12-21:
Hey @staerz ,

I just pushed this feature to master.

Here is an example using your code snippet:
```vhdl
architecture rtl of fifo is

begin

  some_block : block
  begin
    some_entity : entity misc.me
      -- here we go for the next layer of indentation
      generic map (
        first_port => first_port,
        last_port => last_port
      );
      -- here is where the one layer too much of indentation is in my eyes.
      -- because a signal assignment would be here:
    some_signal <= this_signal;
  end block;

end architecture rtl;
```
If I use this configuration file:
```yaml
rule:
  instantiation_034:
    method: entity
indent:
    tokens:
        component_instantiation_statement:
            instantiation_label:
                token : current
                after : current
            semicolon:
                token : current
                after : current
```
...and pass it to VSG using the `-c` option and use the `--fix` option, then VSG produces this output:
```vhdl
architecture rtl of fifo is

begin

  some_block : block is
  begin

    some_entity : entity misc.me
    -- here we go for the next layer of indentation
    generic map (
      first_port => first_port,
      last_port  => last_port
    );

    -- here is where the one layer too much of indentation is in my eyes.
    -- because a signal assignment would be here:
    some_signal <= this_signal;

  end block some_block;

end architecture rtl;
```
I think that matches your desired output.

When you get a chance, could test it out.

Thanks,

--Jeremy

staerz, 2020-12-21:
Hi @jeremiah-c-leary,

that looks like it should do it.

I just pulled the recent master and ran VSG again.

Now I get
```
ERROR: Could not find configuration file: /usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/vhdlFile/indent/indent_config.yaml
```

Is it possible that you forgot to `git add` a file here?

jeremiah-c-leary, 2020-12-21:
looking into it

jeremiah-c-leary, 2020-12-21:
I checked the Travis CI server and the commit for this issue passed, but the next one failed.

jeremiah-c-leary, 2020-12-21:
So I did an install on my machine and I get the same error.

jeremiah-c-leary, 2020-12-21:
so I cloned VSG into a new working copy and did an install.   The local directory has the config file:

```mono
jcleary@DESKTOP-HV9NHA3:~/projects/vsg_debug$ l vsg/vhdlFile/indent/
__init__.py  __pycache__/  indent_config.yaml  set_token_indent.py
```
...but when I install it, the install directory does not have it:
```mono
jcleary@DESKTOP-HV9NHA3:~/projects$ l /usr/local/lib/python3.8/dist-packages/vsg-master-py3.8.egg/vsg/vhdlFile/indent/
__init__.py  __pycache__/  set_token_indent.py
```
There must be something in the setup.py that is preventing the file from being copied.

jeremiah-c-leary, 2020-12-21:
yeah, I check the `build` directory and the file is missing.  What is odd is that the yaml files in the styles directory are there.

I tried changing the extension to `json`, `txt` and `py`.  Only the file ending in `py` was copied.

I do not understand why.

jeremiah-c-leary, 2020-12-21:
Hey @staerz ,

I think I figured it out.  Could you update and verify.

Thanks,

--Jeremy

staerz, 2020-12-21:
Well, if I browse through the installation output, then I actually only see python files being copied (and compiled) over.

Maybe you need to add an extra directive for properly considering the json file?

Note that during installation I get these on `stderr`:
```
/usr/lib/python3.8/site-packages/setuptools/dist.py:480: UserWarning: The version specified ('master') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
  warnings.warn(
warning: no files found matching 'README.md'
```
The first seems fairly expected, but you might want to check the origin of the second.

staerz, 2020-12-21:
Alright, that seems to have improved the situation.

With the configuration that you suggest for my case, I get no complaints - except for a file where I used your default (indented) instantiation.

After I fixed that, I only found lines with the preprocess command (which I didn't indent on purpose).

I would still say that we can close this issue here, as we're discussing preprocessing in #503.

jeremiah-c-leary, 2020-12-21:
Awesome, thx.

closed 2020-12-21 by jeremiah-c-leary