jeremiah-c-leary/vhdl-style-guide #447: errornous group definition for concurrent_006

staerz, 2020-07-20:
**Environment**
VSG 2.0.0, fedora 32

**Describe the bug**
It's all fake news! "Inconsistent alignment of "<=" in group of lines." is reported as a violation of `concurrent_006` with the introduction of VSG 2.0.0 on the following code piece (being the architecture):
```
  combine_data_gen : for i in avst_streams_i'range generate
    avst_stream_o.data(S*(i+1)-1 downto S*i) <= avst_streams_i(i).data;
  end generate;

  -- map the valid bits into one std_logic_vector
  set_valid_gen : for i in avst_streams_i'range generate
    valids(i) <= avst_streams_i(i).valid;
  end generate;

  -- check if they are all up
  avst_stream_o.valid <= allone(valids);
```

Clearly, these are 3 independent groups, but the current configuration doesn't allow to handle that, it seems.

I hope I haven't missed anything for the configuration of VSG 2.0.0!

**To Reproduce**
1. Have an architecture implement some code like above
2. Run VSG
3. See error

**Expected behavior**
I would not expect a violation of `concurrent_006` here.

jeremiah-c-leary, 2020-07-20:
Those should be treated separately.  I will investigate.

jeremiah-c-leary, 2020-12-05:
@staerz ,

I created an example file using your code snippet above and ran it against master.  Master did not throw `concurrent_006`.

Could you try the code against master and let me know if you still see an issue?

Thanks,

--Jeremy

staerz, 2020-12-10:
Confirmed, it doesn't happen with the latest master anymore (2.0.0-457-g30bfcad7). Thanks!

closed 2020-12-10 by staerz