jeremiah-c-leary/vhdl-style-guide #448: Misleading/confusing error message on entity_017

staerz, 2020-07-20:
**Environment**
VSG 2.0.0, fedora 32

**Describe the bug**
The error message for entity_017 is misleading: "Inconsistent alignment of ":" in generic or port declaration of entity."

It is thrown on the following code:
``` 
    --! Global clock
    clk     : in    std_logic;
    --! Global reset
    rst     : in    std_logic;
    --! BCR dummy signal (clk, being not the LHC 40 MHz, but the internal clk_fpga)
    bcr_i   : in    std_logic;
    --! BCID interface created from BCR
    bcid_o  : out   avst_bcid_t
```
Fixing it, leads to that piece of code:
``` 
    --! Global clock
    clk    : in    std_logic;
    --! Global reset
    rst    : in    std_logic;
    --! BCR dummy signal (clk, being not the LHC 40 MHz, but the internal clk_fpga)
    bcr_i  : in    std_logic;
    --! BCID interface created from BCR
    bcid_o : out   avst_bcid_t
```
Note that the difference is 1 space that was removed in front of each single ":" sign.

Previous VSG versions wouldn't bother for how many spaces there are, unless at least the number as configured by "signal_003, spaces" is respected.

Note, that the initial code has the ":" aligned to the next "tab", while apparently VSG 2.0.0 just checks for the longest port name, and adds 1 space. Is that the correct understanding - or would there be room for configuring this? If so, how?

**To Reproduce**
1. Use code as above
3. Run VSG
4. See error

**Expected behavior**
I would expect the error message to be more specific, i.e. point out that there are too many spaces (as the alignment is correct).

jeremiah-c-leary, 2020-12-05:
Hey @staerz,

Somewhere along the way, we changed the alignment base rule to be more configurable and had to make some of the options default.  For example, on master if you check the possible configuration options for `entity_017`, using `vsg -rc entity_017`, you will see:
```json
{
  "rule": {
    "entity_017": {
      "indentSize": 2,
      "phase": 5,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "compact_alignment": true,
      "blank_line_ends_group": true,
      "comment_line_ends_group": true,
      "separate_generic_port_alignment": true
    }
  }
}
```
I believe we changed the `compact_alignment` to true.  This would have forced everything to the left by one column.  To match your previous format, you would need the following configuration:
```json
  {
    "rule": {
      "entity_017": {
        "compact_alignment": false,
        "blank_line_ends_group": false,
        "comment_line_ends_group": false,
        "separate_generic_port_alignment": true
      }
    }
  }
```
The `blank_line_ends_group` would not apply in your snippet though.

Let me know if this answers your question.

Regards,

--Jeremy

staerz, 2020-12-10:
Alright, thanks for the explanation. With the settings as you tell, it works as expected, and also now the text for a violation is much more explanatory. Thanks!

closed 2020-12-10 by staerz