jeremiah-c-leary/vhdl-style-guide #449: Modify architecture_003 to allow comment before architecture

staerz, 2020-07-20:
**Is your feature request related to a problem? Please describe.**
The rule `architecture_003` requires an empty line before the `architecture` keyword.

That is fine, but lacks the option that a comment (i.e. for doxygen or other documentation tools) may be added.

Hence
```vhdl
... some fancy code

--! @brief Some brief description of my_thing 
architecture behavorial of my_thing is
...
```
is a violation of that rule and one would have to make it
```vhdl
... some fancy code

--! @brief Some brief description of my_thing 

architecture behavorial of my_thing is
...
```
which adds an unnecessary blank line.

Note that doxygen comments are very sensitive to blank lines.

Also, the rule should allow any number of lines of comment before the `architecture` keyword.
I.e.
```vhdl
... some fancy code

--! @brief Some brief description of my_thing 
--! @details Here we do something
--! really really important
--! and we must spend
--! many lines of details here
--! as it is so crucial to understand 
--! what we do here
architecture behavorial of my_thing is
...
```

**Describe the solution you'd like**
Allow any number of comments before the `architecture` keyword and then only the empty line.

**Describe alternatives you've considered**
Disabling the rule.

**Additional context**
Doxygen, good quality documentation of code.

Similar comments/feature requests hold for similar rules - should I open dedicated issues?

jeremiah-c-leary, 2020-07-20:
I think this one issue can handle the entity, library, architecture and type rules.

staerz, 2020-07-20:
Alright, then I won't open further issues.

jeremiah-c-leary, 2020-12-05:
Hey @staerz ,

When I was refactoring all the rules for the new parser, I update the rules that check for blank lines to allow for comments.  This is configurable, and you can check the configuration using `vsg -rc architecture_003`:

```mono
jcleary@DESKTOP-HV9NHA3:~/projects/vhdl-style-guide$ vsg -rc architecture_003
{
  "rule": {
    "architecture_003": {
      "indentSize": 2,
      "phase": 3,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "allow_comments": false
    }
  }
}
jcleary@DESKTOP-HV9NHA3:~/projects/vhdl-style-guide$ vsg -rc entity_003
{
  "rule": {
    "entity_003": {
      "indentSize": 2,
      "phase": 3,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "allow_comments": false
    }
  }
}
jcleary@DESKTOP-HV9NHA3:~/projects/vhdl-style-guide$ vsg -rc entity_003
{
  "rule": {
    "entity_003": {
      "indentSize": 2,
      "phase": 3,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "allow_comments": false
    }
  }
}
jcleary@DESKTOP-HV9NHA3:~/projects/vhdl-style-guide$ vsg -rc type_010
{
  "rule": {
    "type_010": {
      "indentSize": 2,
      "phase": 3,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "allow_comments": false
    }
  }
}
```

You just have to set `allow_comments` to `True`.

Could you check on master to see if this is working for you.

Otherwise,  I will plan to close this when I release 3.0.0

Thanks,

--Jeremy

staerz, 2020-12-17:
Hi @jeremiah-c-leary,

works like a charm! Tested with VSG in f1ba134c.

Thanks a lot!

jeremiah-c-leary, 2020-12-17:
awesome,  I will close this issue.

closed 2020-12-17 by jeremiah-c-leary