jeremiah-c-leary/vhdl-style-guide #451: Improve documentation on suffix/prefix rules

staerz, 2020-07-20:
**Is your feature request related to a problem? Please describe.**
The VSG documentation is very good in general.

VSG 2.0.0 introduced the rules for prefixes and suffixes, which, on its own, is a very awaited feature, but its documentation has room for improvement.

**Describe the solution you'd like**
The documentation should be improved listing all rules that have the option of prefix/suffix in [the central documentation of that feature](https://vhdl-style-guide.readthedocs.io/en/2.0.0/configuring.html#configuring-prefix-and-suffix-rules).

That would help a lot to actually find these rules.

Right now, the only way is to go rule by rule and see if it's a prefix/suffix rule... very cumbersome.

**Describe alternatives you've considered**
Not having these rules enabled since they're disabled by default - which makes it even harder to find them.

staerz, 2020-07-20:
PS: I think that's my last comment on VSG 2.0.0 :wink:

jeremiah-c-leary, 2020-07-28:
I wonder if I could auto update the documentation to pull the rules that apply.  Otherwise, maintaining the list will be troublesome.

staerz, 2020-07-28:
So yeah, no clue how you do your documentation. Would it be possible to link pieces by some anchors? Like an include, or a real link? Then you could mark this anchor to the rules that apply - and the rules then just point back to the global rule documentation.

jeremiah-c-leary, 2020-12-06:
Hey @staerz ,

I updated the documentation for the prefix and suffix configuration.

Could you let me know if it is sufficient?

Thanks,

--Jeremy

staerz, 2020-12-17:
Yes, absolutely, very good! Thanks for this improvement!

closed 2020-12-17 by staerz