jeremiah-c-leary/vhdl-style-guide #452: Tiered approach to case rules

Malcolmnixon, 2020-07-27:
**Is your feature request related to a problem? Please describe.**
I'm working on a project where the coding standard specifies keywords in upper-case, and user-identifiers in lower-case.
In order to get this working I've had to build a style-rules.yaml file with an enormous number of case rules.

**Describe the solution you'd like**
The 'generic case' rule sets the default case, which can be overriden by more specific rules. It would be nice to have an intermediate set of case rules:
* keyword_case: 'upper'
* identifier_case: 'lower'

Other case rules would then inherit these values as defaults, and these would inherit from the generic 'case' rule if not provided.

**Describe alternatives you've considered**
Nothing has come to mind.

**Additional context**

jeremiah-c-leary, 2020-07-27:
@Malcolmnixon ,

That sounds like a good idea.

It fits in nicely with the code restructuring underway that would make that fairly easy to do.

jeremiah-c-leary, 2022-01-02:
Hey @Malcolmnixon,

I just merged branch issue-651 to master and will be releasing it as version 3.5.0.  If you get a chance, please try it out and let me know what you think.

Regards,

--Jeremy

closed 2022-01-02 by jeremiah-c-leary