jeremiah-c-leary/vhdl-style-guide #454: (Add option to) shorten VSG output

staerz, 2020-08-03:
**Is your feature request related to a problem? Please describe.**
From VSG 1.11.1 to VSG 2.0.0 the produced output became longer.

In particular when everything goes well, I find it unnecessarily long:

```bash
================================================================================
File:  the_name_of_the_file.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 387
Total Violations:    0
  Error   :     0
  Warning :     0
```

Instead, something like this would fairly be enough:
```bash
File:  the_name_of_the_file.vhd OK (387 rules checked)
```

**Describe the solution you'd like**
In the case that a file is checked successfully (no violations), a simple one-liner would be enough (see above proposal).

**Describe alternatives you've considered**
Piping std_out to nil.

Which does not work either since VSG output doesn't distinguish between stdout and stderr.

**Additional context**
I had expected that VSG would print errors to stderr, but just tried now for the first time and see it doesn't do what I'd expect. Apparently all output is for stdout, no matter if things go right or wrong. (The exit code though is fine.)

jeremiah-c-leary, 2020-08-04:
@staerz ,
Someone else asked for stderr also.

Have you tried the **-of syntastic** option?  The only problem with that is it doesn't report anything if everything passes.

staerz, 2020-08-04:
> @staerz ,
> Someone else asked for stderr also.
> 
> Have you tried the **-of syntastic** option? The only problem with that is it doesn't report anything if everything passes.

I just tried that one (I never did realise this option before).

So yes-ish: the output is shorter, but note that errors are still reported to stdout, not stderr. And as you say, there's no output at all for passing files, which I think is a problem.

So that's not what I'm looking for.

Usually a "--quiet/-q" option would suppress unnecessary output, but by default you'd still get it. (And a "--verbose" option would give you all fancy detailed output that you might need in few cases).

jeremiah-c-leary, 2020-12-05:
what about something like this:

```mono
File:  first.vhd OK (400 rules checked) [Errors: 0] [Warnings: 10] [Guidelines: 0]
File:  second.vhd ERROR (324 rules checked) [Error: 10] [Warnings: 0] [Guidelines: 5]
File:  third.vhd OK (400 rules checked) [Errors: 0] [Warnings: 1] [Guidelines: 3]
File:  fourth.vhd ERROR (120 rules checked) [ERROR: 20] [Warnings: 0] [Guidelines: 0]
```

And the `ERROR` lines would be sent to stderr.

staerz, 2020-12-17:
Yes, that proposed output seems very appropriate and is very nice to parse on with sed or other command line tools.

(Note that I think you have an unintended typo in your last line and meant not to capitalise "ERROR", but write "Error" instead.)

Once you come up with a short report, please make sure to document it and never change it, unless specified in release notes.
Again, I'm very happy that VSG is a command line tool, and parsing its output is essential for us, which relies on a fixed format.

jeremiah-c-leary, 2020-12-17:
> Note that I think you have an unintended typo in your last line and meant not to capitalise "ERROR", but write "Error" instead.

Looks like you are correct.  The output should be:

```mono
File:  first.vhd OK (400 rules checked) [Errors: 0] [Warnings: 10] [Guidelines: 0]
File:  second.vhd ERROR (324 rules checked) [Error: 10] [Warnings: 0] [Guidelines: 5]
File:  third.vhd OK (400 rules checked) [Errors: 0] [Warnings: 1] [Guidelines: 3]
File:  fourth.vhd ERROR (120 rules checked) [Errors: 20] [Warnings: 0] [Guidelines: 0]
```

> Maybe one comment: Usual notification levels would be Fatal, Error, Warning, Info, Debug, so "Guideline" doesn't really fit that,

So the thinking was my thinking:
*  `Error` was something that must be fixed
*  `Warning` was something that should be fixed
* `Guideline` was something that would be nice to fix

Rules like `with_001` could be a `guideline`, at least in my group, to say..."It's not wrong, but in this group we like to use this other language construct."

Maybe there isn't much of difference between `Guideline` and `Warning`?

> parsing its output is essential for us, which relies on a fixed format.

May I ask what you do with the information after you parse it?  I am wondering if the junit or json output would fit your needs.

staerz, 2020-12-17:
Hi @jeremiah-c-leary,
> Maybe there isn't much of difference between Guideline and Warning?

That is exactly my point: if your group doesn't like this construct, then you simply request to not have this, enable the rule, and it's a warning. If it's a warning, the return code should still stay 0, (and I think the output should go to stdout), but at least you can then check for warnings - if you want.

In my group, this wouldn't be an issue at all, so I'll simply disable this rule, no warning, no anything, fine.

Well, I'm using the JUnit output already, or course (super handy, and you remember I even reported the issue on the special html characters...) - but in general, output should remain parsable for any purpose.

jeremiah-c-leary, 2020-12-17:
> That is exactly my point: if your group doesn't like this construct, then you simply request to not have this, enable the rule, and it's a warning. If it's a warning, the return code should still [...]

That makes sense.  I will remove the `Guideline` level.

jeremiah-c-leary, 2020-12-17:
Hey @staerz ,

I just pushed an initial implementation for this issue.  When you get a chance could you check it out.

Thanks,

--Jeremy

staerz, 2020-12-17:
Alright, I saw the relevant commit and tested it, looks very good to me!

Also printing to stdout vs. stderr is done properly.

Thanks! I think you can close the issue.

jeremiah-c-leary, 2020-12-18:
Awesome!!!

closed 2020-12-18 by jeremiah-c-leary