jeremiah-c-leary/vhdl-style-guide #455: Make rule signal_016 compliant with VHDL-2008

staerz, 2020-08-03:
**Environment**
VSG 2.0.0

**Describe the bug**

```bash
  signal_016                | Error      |         48 | ensure signal declaration is on a single line.
Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg==2.0.0', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/site-packages/vsg-2.0.0-py3.8.egg/vsg/__main__.py", line 377, in main
    oRules.report_violations(commandLineArguments.output_format)
  File "/usr/local/lib/python3.8/site-packages/vsg-2.0.0-py3.8.egg/vsg/rule_list.py", line 252, in report_violations
    report.vsg_stdout.print_output(dRunInfo)
  File "/usr/local/lib/python3.8/site-packages/vsg-2.0.0-py3.8.egg/vsg/report/vsg_stdout.py", line 14, in print_output
    print_violations(dRunInfo)
  File "/usr/local/lib/python3.8/site-packages/vsg-2.0.0-py3.8.egg/vsg/report/vsg_stdout.py", line 97, in print_violations
    sOutputString += dViolation['solution']
TypeError: can only concatenate str (not "NoneType") to str
```

VSG utterly crashes (see above error output) upon the following code:
```vhdl
  signal avst_packet_o : t_avst_packet(
    data(G_AVST_DATA3_W-1 downto 0),
    empty(AVST_EMPTY_W-1 downto 0),
    error(0 downto 0)
  );
  signal avst_packets_i : t_avst_packets(3 downto 0)(
    data(G_AVST_DATA3_W-1 downto 0),
    empty(AVST_EMPTY_W-1 downto 0),
    error(0 downto 0)
  );
```

with and underlying
```vhdl
  type t_avst_packet is record
    data:   std_logic_vector;
    valid:  std_logic;
    sop:    std_logic;
    eop:    std_logic;
    empty:  std_logic_vector;
    error:  std_logic_vector;
  end record;

  type t_avst_packets is array (natural range <>) of t_avst_packet;
```

Note that the above is VHDL-2008 compliant, but not VHDL-1993:
It is to declare a signal (constrained) from an unconstrained record type.

**To Reproduce**
Steps to reproduce the behavior:
1. Run VSG 2.0.0 on the above code snippet

**Expected behavior**
No crash of VSG.

**Additional context**
VHDL-2008

jeremiah-c-leary, 2020-08-04:
Hey @staerz ,

I am working my way through my new parser and will catch this as part of that effort.

staerz, 2020-08-04:
OK, great. Please just make sure it would parse the example above without a false complaint :wink:

staerz, 2020-08-07:
I found another example where VSG (now I'm back to 1.11.1) would make a false complaint:
```vhdl
  signal byteen_validity_a : boolean :=
    ite(BYTEEN_WIDTH /= 1,
      ite(D_WIDTH_A mod BYTEEN_WIDTH = 0,
        BYTE_SIZE_EFFECTIVE = 5 or BYTE_SIZE_EFFECTIVE = 8 or BYTE_SIZE_EFFECTIVE = 9 or BYTE_SIZE_EFFECTIVE = 10,
        false
      ),
      true
    );
```

Here, `ite` is simply an **i**f-**t**hen-**e**lse function that takes 3 arguments: `ite(<condition>,<return on true>,<return on false>)` - and you could, of course, imagine any kind of function here.

On the syntax, it is debatable, maybe it should better be
```vhdl
  signal byteen_validity_a : boolean := ite(
    BYTEEN_WIDTH /= 1,
    ite(D_WIDTH_A mod BYTEEN_WIDTH = 0,
      BYTE_SIZE_EFFECTIVE = 5 or BYTE_SIZE_EFFECTIVE = 8 or BYTE_SIZE_EFFECTIVE = 9 or BYTE_SIZE_EFFECTIVE = 10,
      false
    ),
    true
  );
```
(so one indentation less, I'd say that's better as it goes with the parenthesis ... and the first argument could be on the same line, I'd say - I used mixed syntax now: the first `ite` has the first argument in the new line, the second one in the same ...)

jeremiah-c-leary, 2020-09-23:
Hey @staerz ,

I took your examples and put them into a file:

[image]

and then ran VSG on it:

[image]

So that is some progress.  As much as a pain it was to implement the new parser, it is much more robust than what I had before.

staerz, 2020-09-23:
Hi @jeremiah-c-leary,

alright, that's very good news!

I'll be happily waiting for you to release the new version then!

jeremiah-c-leary, 2020-12-04:
Hey @staerz ,

I ran the code again through the latest on master and it did not fail.  However, I did have to disable rule signal_016, otherwise it takes the multi line signal assignments and stick them all on the same line.

Could you run against master I and let me know if the issue is resolved?

Thanks,

--Jeremy

staerz, 2020-12-17:
Hi @jeremiah-c-leary,

I confirm that VSG doesn't crash when encountering this code anymore.

Though I still don't like that `signal_016` is complaining here, but that's a different story.

I leave you the pleasure of closing this issue.

jeremiah-c-leary, 2020-12-17:
Awesome, thx.

closed 2020-12-17 by jeremiah-c-leary