jeremiah-c-leary/vhdl-style-guide #456: Allow to enforce to omit certain optional VHDL keywords

staerz, 2020-08-03:
**Is your feature request related to a problem? Please describe.**
For consistency, instead of simply disabling rules like `architecture_010`, `architecture_011` or `process_012` on **optional** VHDL keywords, names or labels, it would be appreciated to be able to force to actually omit them.

This may be desired for a certain coding style.

**Describe the solution you'd like**
A possible solution would be to add an attribute (e.g. "enforce") for the rules in question in case they are disabled:
- If a rule is disabled, but **not enforced**, the entire rule check is not done (so it's a don't care)
- if a rule is disabled but **enforced**, that means that the optional keyword (name or label) **must not** appear - and a fix would be to remove it

In case a rule is enabled, the attribute "enforce" would be insignificant (and simply apply the rule as currently defined).

**Describe alternatives you've considered**
An alternative solution would be to introduce exclusive rules, but that would be impossible to steer as exclusive rules must never be enabled at the same time - how could one check this (and it would make everything utterly complicated).

The ultimate alternative is to simply ignore the undesired optional keywords in case the rules are disabled.
There's nothing that one could do in this case (about enforcing a certain style).

**Additional context**
Consistency matters.

jeremiah-c-leary, 2020-08-04:
Hey @staerz ,

I could add an option for those rules that **add** the optional parameters to make those rules **remove** them.  Then there is a single rule for that optional argument.

staerz, 2020-08-04:
That is basically what I am proposing, having used different words.

jeremiah-c-leary, 2020-12-12:
Hey @staerz ,

I have implemented this feature.

I will close this ticket when I release 3.0.0 if you do not have enough time to validate it.

Thanks,

--Jeremy

staerz, 2020-12-17:
Hi @jeremiah-c-leary,

that is great!

Note that I found yet another of these documentation bugs: In the list of rules that have this feature of "add" or "remove", the links are again broken: the links use "underscore", while it should be "dash". I think that's true for any lists there in the documentation (so also [the prefixes and suffixes for example](https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#rules-enforcing-prefixes-and-suffixes) - seems to be a general readthedocs thing ...)

I have tested it for `architecture_010` and it works like a charm (both adding and removing).

jeremiah-c-leary, 2020-12-17:
Hey @staerz ,

The link had an extra extension in the filename.  It should be fixed now.

Thanks,

--Jeremy

jeremiah-c-leary, 2020-12-28:
Hey @staerz,

I am closing this with the update to the documentation.

Thanks,

--Jeremy

closed 2020-12-28 by jeremiah-c-leary