jeremiah-c-leary/vhdl-style-guide #457: Add rules for the block keyword (environment)

staerz, 2020-08-03:
**Is your feature request related to a problem? Please describe.**
VSG 2.0.0 currently doesn't have any rules for the `block` keyword.

**Describe the solution you'd like**
Implement rules similar to those of `architecture`:
- naming (upper/lower case, label)
- (optional) `is` keyword
- begin/end (with label)
- ...

**Describe alternatives you've considered**
There is no alternative.

**Additional context**
I was wondering why my blocks can do what they want until I realised that VSG doesn't give a sh*t.

jeremiah-c-leary, 2020-08-04:
> I was wondering why my blocks can do what they want until I realised that VSG doesn't give a sh*t.

lol, that is true.  I have never used them.

As I rework the parser I will add rules for them.

jeremiah-c-leary, 2020-12-10:
My initial set of rules have been documented.  I will work on implementing them tomorrow.

jeremiah-c-leary, 2020-12-10:
Hey @staerz ,

I just added 29 block rules.  Could you look them over and test them out.

Thanks,

--Jeremy

staerz, 2020-12-17:
Hi @jeremiah-c-leary,

is it possible that the block rules are behind a feature flag or so?

I'm running on f1ba134c and with a `vsg <myconfig> -oc config.vsg` I find no rule called `block_*` - is that expected?

Also I added an obvious violation of `block_001` and VSG didn't complain. Similarly, when trying to configure `block_002`, I get a complaint by VSG that the rule wouldn't exist.

In which trap did I run?

jeremiah-c-leary, 2020-12-17:
hmm...I forgot to add it to the list of rules to use.   I just pushed an update that should include it.

jeremiah-c-leary, 2020-12-28:
Hey @staerz,

Wondering is this issue could be closed.

Thanks,

--Jeremy

staerz, 2020-12-30:
Yes, I'd say you can close this one here, block rules now exist. If I ever spot any problem with them, I'll report in a dedicated issue.

jeremiah-c-leary, 2020-12-30:
awesome, thx.

--Jeremy

closed 2020-12-30 by jeremiah-c-leary