jeremiah-c-leary/vhdl-style-guide #458: vsg is installed but can not use

cuijialang, 2020-08-07:
**What is your question?**
Ask a question and provide as much relevant information as possible.
vsg is installed but can not use , report below:
Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg==2.0.0', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.6/site-packages/vsg-2.0.0-py3.6.egg/vsg/__main__.py", line 327, in main
    commandLineArguments = parse_command_line_arguments()
  File "/usr/local/lib/python3.6/site-packages/vsg-2.0.0-py3.6.egg/vsg/__main__.py", line 37, in parse_command_line_arguments
    parser.add_argument('--style', action='store', default=None, choices=get_predefined_styles(), help='Use predefined style')
  File "/usr/local/lib/python3.6/site-packages/vsg-2.0.0-py3.6.egg/vsg/__main__.py", line 62, in get_predefined_styles
    tempConfiguration = yaml.full_load(yaml_file)
AttributeError: module 'yaml' has no attribute 'full_load'

jeremiah-c-leary, 2020-08-07:
Hey @cuijialang ,

Check which version of PyYAML is installed.  I remember having to upgrade my version.

cuijialang, 2020-08-07:
hi ,
my version of PyYAML is PyYAML-3.12-py3.6.egg

cuijialang, 2020-08-08:
but how to continue
cuijialang@localhost vhdl-style-guide-master]$ pip3 install vsg
Requirement already satisfied: vsg in /usr/local/lib/python3.6/site-packages/vsg-2.0.0-py3.6.egg
Requirement already satisfied: PyYAML in /usr/lib64/python3.6/site-packages (from vsg)

cuijialang, 2020-08-08:
i fix the question use below changes:
modify code __main__.py :line 62 and line 88
tempConfiguration = yaml.full_load(yaml_file)-->tempConfiguration = yaml.safe_load(yaml_file)

closed 2020-08-08 by cuijialang

jeremiah-c-leary, 2020-08-08:
Hey @cuijialang ,

I checked my linux install and I am using version 5.1.

staerz, 2020-09-23:
Hi @jeremiah-c-leary,

please note that I just ran into the same issue.

Doing a `pip3 install PyYAML` (pulling in PyYAML-5.3.1) solved the problem.
Is it possible that the installation procedure for VSG is incomplete in that respect (I have no idea of how that's done anyway, so just asking a question of sth. I have no clue about ...)?

jeremiah-c-leary, 2020-09-25:
I wonder if there is a way to specify a minimum release for PyYAML in my setup file.

jeremiah-c-leary, 2020-09-25:
Looks like I can.  I added it to my master.  It will hit on my next push.  It may be awhile.  I just finished re-writting the keyword alignment.  That is probably one of the hardest rules I have.