jeremiah-c-leary/vhdl-style-guide #459: Fix instantiation_011 parsing type conversion using a constant

staerz, 2020-08-10:
**Environment**
VSG 1.11.1 (but likely also 2.0.0)

**Describe the bug**
`instantiation_011: 50 : Change port name to lowercase.` is thrown although the port name itself is in lower case in case given below:

```vhdl
uut: entity deep_testing.ram_ctrl
  generic map (
    RAM_N       => RAM_N
  )
  port map (
    clk           => clk,
    srst          => srst_i,
    valid_i       => valid_i,
    din           => std_logic_vector(to_unsigned(counter_d, D_WIDTH_A)),
    valid_o       => valid_o,
    dout          => dout
  );
```

Note that `D_WIDTH_A` is an actual constant defined in a library (package) included earlier.

The code given is simply using the type conversion in a port map, and `to_unsigned` simply requires the length of the signal which is usually given via a constant.

One can imagine other functions of any combination of signals or constants as well and I'd say it's impossible to make a rule check on anything more complex than a simple word (after the `=>` port map operator).

**To Reproduce**
1. Run VSG on the code snippet
2. See error

**Expected behavior**
VSG should not throw this error here.

**Proposed solution**
Restrict the check for upper/lower case to the left side of the `=>` port map operator.
For the right side of it, restrict the check to a simple word (`\w` in regex): If it is any more complex than it's a function, maybe one can still check that the function name matches the rule for functions upper or lower case, but everything beyond (= arguments of that function) should be ignored. (BTW: there seems not to be a rule that check on the casing of the function name itself - did I miss anything?)

As said above, I cannot imagine a fail-proof implementation of also parsing the function arguments right (unless one actually parses the function itself for its arguments - but that function may be defined in any file which is not necessarily the current file).

staerz, 2020-09-21:
Hi @jeremiah-c-leary,

it has been a while since there hasn't been any update on the side of VSG.

I'm wondering when you'd be able to fix a few problems that I'm currently struggling with (and reported).

Please let me know about any tentative date or plans.

Thanks

jeremiah-c-leary, 2020-09-23:
Hey @staerz,
I have been hunkered down writing a new parser for VSG that can handle these cases and not throw false alarms.  The regular expression method I was using was not robust enough to handle complex cases.  Here is my current schedule for VSG:

[image]

The rule transition phase is underway and working my way through the architecture rules.     
 I have some base rules defined already like case checks, that make rule creation pretty trivial.  It takes longer for me to create the tests than to convert the rule.

There are about 350 rules.  I we estimate 10 min per rule, that would be 350 * 10 min = 3500 min / 60 min ~= 58 hours.  Let's say I can devote 4 hours a day, that makes 25 days. 
 So the end of October seems reasonable.

If this issue is holding you up I could look at what it could take to fix.  Just let me know.

Regards,

--Jeremy

staerz, 2020-09-23:
Hi @jeremiah-c-leary,

thanks very much for that immediate reply. That helps me plan on my side!

No, please don't waste any of your precious time for finding an interim solution, please just pursue your plan of reworking your parser (and let me know if you want me to test anything ...). I will hence go ahead and deal with VSG 2.0.0 as it is right now. Knowing that some (tremendous) improvement is on its way is enough for me.

Again, thanks a lot!

staerz, 2020-11-23:
Hi @jeremiah-c-leary,

how are things, does the migration go according to plan?

I'm just asking to get a feeling such that I can adopt accordingly for our project. This is not to rush you in any way.

Thanks

jeremiah-c-leary, 2020-11-23:
Hey @staerz,

I completed all the rule conversions and a phase of hardening such that all of my tests are now passing.  The tests include some code examples.  I am currently pulling example code from open cores and running against it.

I have found a couple of minor issues:

* ISO 8859-1 encoding support
* a rule with a missing solution

I am trying to find very complex code to run against to ensure my parser doesn't choke.  The new parser should handle nested blocks and generates, etc... 

If this phase goes well, I hope to start working on issues at the beginning of December instead of January.

If you want, you could pull down master and see if it resolves some of the issues you have reported.

staerz, 2020-11-23:
Hi @jeremiah-c-leary,

thanks for the update. That sounds very good.

And yes, I can certainly run VSG in the current master over our code base, there are several 10k of lines to check :wink:
And I'll see (and report back) if some of the recent issues (with VSG 2.0.0) that I reported would be fixed by the new version or not.

jeremiah-c-leary, 2020-12-04:
Hey @staerz ,

I was able to improve the performance significantly, although it is a little slower than version 2.   The parser is more robust than the regular expressions I was using.

So I copied the example code above, wrapped it in an architecture and ran it against master.  Good news, master did not throw the instantiation_011 error:

```mono
jcleary@DESKTOP-HV9NHA3:~/projects/vhdl-style-guide/issue_validation/issue_459$ cp example.orig.vhd example.fixed.vhd;vsg -f example.fixed.vhd
================================================================================
File:  example.fixed.vhd
================================================================================
Phase 2 of 7... Reporting
Total Rules Checked: 169
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  instantiation_003         | Error      |          7 | Ensure only one space before the :.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
jcleary@DESKTOP-HV9NHA3:~/projects/vhdl-style-guide/issue_validation/issue_459$ cp example.orig.vhd example.fixed.vhd;vsg -f example.fixed.vhd --fix
================================================================================
File:  example.fixed.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 431
Total Violations:    0
  Error   :     0
  Warning :     0
```

I believe the comma in the function call was confusing version 2.  It was considering the comma to be the separator in an association_list.  The new parser is able to handle this case correctly.

Could you run with the latest on master and let me know if it still has an issue?

Thanks,

--Jeremy

staerz, 2020-12-09:
I can confirm that it doesn't happen anymore with latest master (2.0.0-457-g30bfcad7).

closed 2020-12-09 by staerz